cncjs/cncjs #662: automate center finding

slimline33, 2020-11-25:
Some automated center finding would be a nice to have option :) I think it is just a macro?

https://youtu.be/aT_9LtDPnKw

neilferreri, 2020-11-25:
Center finding for a hole?
Or just probing for a workpiece corner?
Did you mean something different?
https://github.com/cncjs/CNCjs-Macros

slimline33, 2020-11-25:
Sorry, not properly explained.

I mean there is already the probe widget for XYZ. I think it would be good if there is the possibility to use two or at least three side buttons cncjs to set the center of the workpiece to centered center.

Like in the Video. Just tap 3 sides with your probe and cnjs would Set workpiece Zero in the exact middlepoint