cncjs/cncjs #664: Wrong regex in state parser for Smoothieware

sigxcpu76, 2020-11-30:
#### Description

Smoothieware returns the state prefixed with `GC:` just like GRBL 1.1. CNCjs does not take this into account and it pollutes the console with the state line (i assume because it is not parsed).

I have changed the `SmoothieLineParserResultParserState` function to match that `GC:` prefix and all seems ok now.

`line.match(/^\[GC\:((?:[a-zA-Z][0-9]+(?:\.[0-9]*)?\s*)+)\]$/);`

#### Versions

- CNCjs: 1.9.22
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [x ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [ x] Linux

mkotyk, 2020-12-03:
Hey @sigxcpu76 I encountered the same issue and made a pull request for a fix.  https://github.com/cncjs/cncjs/pull/661  Not sure if it will get merged or not.  It's pretty straight forward.

emcniece, 2022-03-26:
Hi! Is this issue still happening now that https://github.com/cncjs/cncjs/pull/661 is merged?

sigxcpu76, 2022-03-27:
I am not using Smoothieware anymore, so I cannot check. Switched back to GRBL b/c SW does not have a "realtime" stop/emergency.