cncjs/cncjs #669: Lost Widget/Workspace Settings?

michaelquigley, 2020-12-16:
#### Description

I'm not sure this is an issue. I may just be misunderstanding how CNCjs works.

I've got CNCjs up and working on a Raspberry Pi 3B+. I had it working on Windows 10 before I went ahead and got the Pi. My installation has a user account set up.

Every time I re-arrange and disable the unnecessary widgets in the workspace, the configuration is lost when I leave CNCjs and come back. I'd really like to not have to re-configure the workspace every time I want to use CNCjs.

I searched the issues, and I looked through the docs and didn't find anything that seemed relevant. What am I missing?

#### Versions

- CNCjs: 1.9.22
- Node.js: 8.17.0
- NPM: 

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

slimline33, 2020-12-16:
do you have cookies enabled or any script blocker?

michaelquigley, 2020-12-16:
No. In fact I've been very careful to turn all of those kinds of protections off for the CNCjs address in my browser (Firefox).

Am I misunderstanding how this works? I would expect to be able to log into my CNCjs workspace, adjust all of the widgets to where I want them and which I want to see... and then I would expect to be able to log in from another browser on another machine and have that workspace configuration be present... i.e. I'm expecting CNCjs to preserve those settings in my account server-side.

Is that not how this is supposed to work?

TJ-Houghton, 2021-02-06:
I am having the same problem.  Macros saved fine, but my widget setup returns to default everytime I restart my pi.

Versions:

CNCjs: 1.9.22
- Node.js: 10.23.2
- NPM: 6.14.10
- CNC Controller

Grbl
Hardware

Raspberry Pi

closed 2026-03-13 by cheton