cncjs/cncjs #670: The minimum is limited to the laser widget

komorebi-nan, 2020-12-19:
#### Description

Hi, I am so happy to see such a fruitful project! But I am a little confused in using it.
I have a high power laser transmitter, but sometimes the power is too high for the material. Even ten percent of the power. So I would like to be able to adjust the power below 10 percent in the laser widget.
It's not really a problem, it's just a special case of mine. I hope I can get help. Thank you very much.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2020-12-19:
The laser widget is not meant to be the primary means of controlling the laser power.  Primary control is via the "S" word in a GCode command.  The widget supplies an "override" to let you make adjustments to the power specified by the GCode program.

This is the same model that is used for spindle speed overrides and feedrate overrides; the GCode program establishes the baseline value.  The user interface then lets you make moderate adjustments dynamically.

closed 2020-12-19 by MitchBradley

komorebi-nan, 2020-12-21:
I get it. thank you!