cncjs/cncjs #671: Confused about Tool Change Macro

water-rhatt, 2020-12-23:
#### Description

I have read through the tool change Wiki and all the open issues I could find, but I am still confused about the tool change macro. Here is what I think it does:

Probe new tool
Set WCS Z=0 at the probe location less the plate thickness.

That would seem great if my WCS was set at my machine's table, but I rarely do that. I usually have set the Z=0 at the stock top. Obviously, once I have started milling I can almost never re-zero to the top of stock that isn't flat anymore.

I've used "semi-automatic" tool change macros on Mach3 and UCCNC. There is always an initial calibration probe that first determines the difference between the desired z=0 and the tool change touch plate. 

To make this concrete, imagine I probe stock with any random tool (or a dedicated probe) and establish my z=0 at machine z=-500. With the same probe I probe the touch plate at machine z=-1000. The difference between the two is 500mm.

So now I change tools. I can't probe the stock anymore, but I can probe the touch plate.  Let's assume that the touch plate now probes at machine z=-900.  My new tool is 100mm longer than the old tool. I don't want to set my WCS z=-900, I want to adjust it upwards from z=-500 to z=-400.

The macro laid out in the wiki doesn't seem to do this. Am I missing something? Is there a separate calibration macro? Am I misunderstanding how the macro works?

#### Versions

- CNCjs: 1.9.22
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [x] Linux

neilferreri, 2020-12-23:
Have you looked at my macros here?
https://github.com/cncjs/CNCjs-Macros/tree/master/Initial%20%26%20New%20Tool

water-rhatt, 2020-12-24:
No, I hadn't.  Seems like you have two different macros, as I would have expected. I will look through them now, but I think that is exactly what I was looking for. I feel like the wiki should be updated to at least address this issue. It must be common, and the way it is left me confused.

water-rhatt, 2020-12-24:
First question: what is the "global.state" namespace you are using? Does that have pre-defined values that persist or can be accessed, or is that your own naming convention?

neilferreri, 2020-12-24:
Those are global variables that persist across macros, but they will not persist across instances of CNCjs.
Run the original tool macro anytime you need to set a reference. Then run the new tool macro as many times as you have tool changes. 
They work by adjusting your work zero (rather than a G43.1 TLO), so that will persist across machine power cycled and resets.

neilferreri, 2020-12-28:
@water-rhatt Did you get the Tool Change working?

water-rhatt, 2020-12-28:
I’m still trying to implement it, but not because these macros don’t do the trick. I’m trying to move from a Windows/mach3 environment to a Linux/CNCJS/grbl environments. 

But, to recap these macros:

First it sets wcsZ0. Then it moves to the tool probe location and probes. By definition the new posz is the negative of the difference in machine coordinates between wcsz0 and the tool touch plate. The macro saves the difference in a user parameter that persists across different programs (but not if you shut down CNCJS). 

Next, upon reading a M6, it saves its current state and travels to the tool change location. It probes with the new tool. With a G10 it sets the wcsz0 to the tool touch plate plus the stored parameter. This should adjust the wcsz0 in machine coordinates up or down depending on the difference in the new tool length. 

One last question, is there a line of gcode I can insert into my file headers to trigger the init macro?

neilferreri, 2020-12-29:
Mostly.
Z0 is set independently. The tool change macros don't set it.
If your cam program adds an M6 automatically, you can run the macro, but it will not run on its own. You can still use the tool change across jobs, so some will separate jobs by tool and still use the macros.

closed 2026-03-13 by cheton