cncjs/cncjs #672: HTTPS support

antonok-edm, 2020-12-25:
### Description

It would be great if CNCjs would support TLS / SSL / HTTPS, at least with a built-in self-signed certificate and ideally with support for custom certificates as well. I couldn't find any documentation for it anywhere, so I assume that support doesn't currently exist.

### Versions

- CNCjs: 1.9.x
- Node.js: 10.x
- NPM: 5.x

### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [x] Marlin

### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

tillig, 2021-01-04:
This is somewhat interesting to me, too, so I did some research on it. It may not be super straightforward to add.

The main server entrypoint [in `server-cli.js`](https://github.com/cncjs/cncjs/blob/35e355e00112b10888bb3fad184641823baf01cf/src/server-cli.js#L95) eventually [starts the web server listening via a custom `webappengine` package](https://github.com/cncjs/cncjs/blob/35e355e00112b10888bb3fad184641823baf01cf/src/server/index.js#L235) that is owned by one of the contributors here. That package [is tied to the Node.js `http` module](https://github.com/cheton/webappengine/blob/9e8ef2d83d10664b026cf1df240adbbb5458e696/src/app/server.js#L5) ([docs here](https://nodejs.org/api/http.html)) and has no reference to [the Node.js `https` module](https://nodejs.org/api/https.html).

I gather to make the changes required in cncjs to support HTTPS that [the `webappengine` package](https://github.com/cheton/webappengine) would need to add HTTPS support first.

Looking at what `webappengine` does, I might even recommend using a more widely supported server package like expressjs, though [there's still work to do to support HTTPS](http://expressjs.com/en/4x/api.html#app.listen) with that.

TobyGWilliams, 2021-01-04:
What would be the benefit to the user of supporting https? The certificates would have to be self signed and the crucially be manually trusted by each client (this gets trickier on tablets) or accepted every time.

Simplest way to add it would be up put a nginx proxy in front of the application and serve that on 443.

antonok-edm, 2021-01-04:
@TobyGWilliams This is not an exhaustive list by any means, but here's a few arguments in favor:
- Privacy by default; users on shared networks should be comfortable manufacturing their own personal items and gifts without fear of being snooped on by others in their home, apartment, etc.
- TLS is a requirement for proper user management, which has proved very useful for Octoprint installations in makerspaces.
- Some less reputable IoT manufacturers may subsidize their products by aggregating data collected through packet sniffing on internal networks, or be vulnerable to attackers which do the same. In these situations, the less that can be sniffed, the better.
- Browsers are increasingly enforcing restrictions on the web APIs available for use from HTTP origins.
- Browsers display the "Not secure" warning in the URL bar for HTTP origins, which is poor UX for new users.

emcniece, 2022-03-26:
While it would be possible for CNCjs to support this, it's also somewhat outside of the scope of the project. Often we see certificate management provided by external services, load balancers like [Traefik](https://traefik.io/) or platforms like [Kubernetes](https://kubernetes.io/docs/tasks/tls/managing-tls-in-a-cluster/).

Certificate management is greatly impacted by the surrounding network configuration. Given the great many deploy methods for this project, (NPM package, Docker, Electron app, Raspbian image, etc) and any number of network configurations (firewalls, bridge adapters, subnets, OS variants, ) implementing anything beyond self-signed certificates would introduce many considerations, and undoubtedly raise many questions. Browsers will not trust self-signed certificates anyway.

That said, we could certainly document some examples using ACME or a reverse proxy - labelling as such and adding to the queue 👍

risacher, 2022-05-22:
I'm curious if the privacy use case raised by @antonok-edm really affects anyone in practice.  Loopback network connections are not snoopable from a LAN, so if people access CNCjs from 'http://127.0.1.1:8000', it's already private.  (But users might have a CNC-server similar to octoprint.) I thought https would be necessary to use the Serial API (https://wicg.github.io/serial/) which is only available from secure contexts, but I since learned that http://localhost is considered a secure context.