cncjs/cncjs #673: Uploading large files >40 MB

slimline33, 2020-12-25:
#### Description

Currently I am trying to laser files created by LASERGRBL. The problem is that files larger than 40 MB are created (without comments).

Unfortunately the CNCJS instance crashes then. 

Is there no solution for this? I know that CNCJS re-parses the files. 

Is it possible to add a routine that recognizes that files larger than 10 MB are loaded and not visualized?

#### Versions

- CNCjs: 1.9.22
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM

#### CNC Controller

- [x] Grbl

#### Hardware

- [x] Raspberry Pi

#### Operating System

- [x] Linux

TJ-Houghton, 2021-02-06:
I am having the same problem.  My file is only 11MB.  I find that the problem only occurs when running cncjs GUI from the PI.  If I open CNCjs from a desktop computer (and using the CNCjs server on the pi), it appears to run fine, however, my computer is on the other side of the house from where my CNC router is, so it's not a practical solution to control my router from there.

Loading a 6MB file seems to work fine.

Versions:
- CNCjs: 1.9.22
- Node.js: 10.23.2
- NPM: 6.14.10

CNC Controller

- Grbl

Hardware
- Raspberry Pi

MitchBradley, 2021-02-07:
CNCjs is written using "big machine" programming techniques that use a lot of memory, which is why it has problems with large files on Raspberry Pi.  It is possible that a Raspberry Pi 4 with its 2GB of memory might work better.

For the person who has success with a desktop computer, you might consider buying an low-end laptop or a tablet to live next to the router. The cncjs-shopfloor-tablet UI can work with tablets costing less than $100.  Used 10" tablets go for about $100, 8" for about $50, and 7" for about $40. If you mount it on an arm attached to the wall, you can position it at a convenient working location out of the way of dust.

If you need to use the full UI (I hardly ever need it for job running), then an inexpensive 12" laptop is a good control surface.  Lenovo Twist, with touchscreen and keyboard so it could run the tablet UI and also the main UI, goes for about $150 on ebay.

TJ-Houghton, 2021-02-07:
Mitch, Thanks.  I am running a Pi 4B with 4GB of RAM.  Do you think getting the 8GB board would solve this problem?  I bought the Pi because I wanted to run cncjs standalone on the PI because my 8 year old laptop was having problems (both wtih CNCjs and Carbide Motion).  I would ideally like to just use my pi and not have to buy a separate computer to send gcode to PI.

slimline33, 2021-02-07:
No. I already tried it with it. There is a other problem with it. I also tried it on a bigger windows Maschine....

MitchBradley, 2021-02-07:
I do not have a Pi4 of any size so I cannot say.  Guessing in situations like this is not much use.  There have been many reports of CNCjs having problems with files larger than a few megabytes.  In some cases, files as small as 4 MB have caused problems.

karoria, 2021-02-08:
I am already running a rpi4 8gb with cncjs. I found that a file successfully loads if it is up to 6MB. Above that, you can't guarantee. So, I think it is not RAM but processor as far as cncjs and rpi combo is considered. This is due to a programming style where whole file needs to be fetched before starting that program. That is also required to generate graphics for that gcode. Fetching a file and generating graphics involves good amount of processor use where rpi falls short. Another issue with cncjs is its file fetching is hard bound with graphics generation. So despite of you keep 3d graphics off, it actually generates graphics in the background which consumes a lot of processing power. I have requested to isolate graphics from file loading to @cheton before. But I understand that it needs a lot of code refactoring and testing afterwards.
So, for time being, i suggest to use bcnc with rpi4 if you run large files regularly. I have tested bcnc successfully running a 36MB program with my pri4 8gb. Above that I have not tested. Consider disabling its graphics.
Another suggestion is to use a standard laptop/PC and stay with cncjs. Here, the file size limit depends on your hardware and I don't really know what size it usually supports.

karoria, 2021-02-08:
I forgot to mention, you can also use grbl_esp32. That way, you will have better step generation as well as its integrated webui is just awesome for large files running. It just doesn't read the whole file and reads only small portion of file and sends to port line by line. In software, it is called "file streaming". And I must say, it is best server using very limited resources of controller. I have tested that 36MB file succcessfully with this interface also. @MitchBradley is active developer in that project.

TJ-Houghton, 2021-02-08:
@karoria  Thanks for the info.  I'll take a look at them.  I moved to cncjs because it could do manual tool changes and TLO calcs (using a macro) and I didn't have to postprocess each tool as a separate file.  But with the file size restrictions, I am back to post processing each tool as a separate file to keep the file size down :)

slimline33, 2021-02-22:
OK. Wenn es wirklich so ein großes Problem ist große Dateien zu verarbeiten, könnte man vielleicht auch über ein G-CODE Splitting nachdenken. Vielleicht eine Automatisierung welche Dateien erkennt  größer als 5 MB diese dann in mehr einzel dateien  splitten.

like https://github.com/spresse1/gcode-splitter

TJ-Houghton, 2021-02-28:
Christophe,  While not ideal, it may be the way I need to go.  I have been trying other gcode senders, but they all seem to have their issues.  UGS sender can't deal with M6 tool changes, bCNC screen is not optimized for a 10" touchscreen, and I haven't been able to figure out how to resume a file after an M6 command is encountered.  Chilipepper runs very sluggishly on my RPI.  I really like the layout of cncjs, I just wish it could handle larger files.

TJ-Houghton, 2021-02-28:
@karoria  - Can the esp32 webui be used with a stock Shapeoko GRBL controller?  I am not really interested in changing controllers right now.

MitchBradley, 2021-02-28:
esp3d_webui needs a controller that can talk to the network.  GRBL running on an AVR Arduino does not have a network interface.

karoria, 2021-02-28:
> @karoria - Can the esp32 webui be used with a stock Shapeoko GRBL controller? I am not really interested in changing controllers right now.

That webui is embedded in grbl_esp32 firmware which requires esp32 specific hardware resources as mentioned by Mitch.

slimline33, 2023-07-22:
Any news here?

karoria, 2023-07-23:
It looks like cncjs is not in active state to resolve old issues. For the persons who had discussed here, I have some new recommendations:
1. For legacy grbl controllers and may be few new varients of grbl, "gSender" is javascript based cross platform sender with good capabilities. Its team is really active. It was born as an extension of cncjs only and today it serves as good sender.
2. Those who want to level up their cnc control, should consider FluidNC or grblHAL. They both support webUI. For most applications, webUI is enough for a robust working machine.

slimline33, 2023-07-23:
thank you @karoria for your tips with the gcode sender!

I like cncjs. It has a smooth ui and it runs perfectly. But today after reading your message here I tried to start cncjs from a virtual machine. I plugged a esp32 fluidnc controller and upload a gcode file wit 66MB. It need time for loading but no error!

Also there is a problem with cncjs and raspberry. Now I think I throw away the raspberry and use the VM instead of it. I need just a longer USB cable and a USB UART bridge.