cncjs/cncjs #675: z axis dropping bellow the limit

thiaggo, 2021-01-05:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ x] Windows
- [ ] Mac
- [ ] Linux

hi  i have been having trouble with my cnc machine  and i dont know how to fix it,   
i built a nikoden bartinik dremmel cnc machine,   and it worked,  for a couple of months i been having this weird problem where in the middle of the job the z axis dorps bellow the limit and ruins it all, and the machine keep going, usually breaking the bit 
i dont know what to do nor what is the problem
i suspect it may have to do with a lost conection from the pc, or maybe the cncjs got in sleep mode?( im always at my computer when it happens, watching a movie or something.  the pc never got turned off or got in sleep mode when this error happined,   so i dont really know what to do nor how to fix it
i have already set the $1=255
please help   thank you for your time

maroussel, 2021-01-05:
Hello @thiaggo. Got a similar issue on different machine and setup(Workbee on G2Core). It's probably not the same thing but I did various things to investigate so I might be able to help. In my case the axis were refusing to change direction randomly. For example, I was lowering Z axis to carve something and when a command was sent to raise it, something it was actually going even lower by the amount it was supposed to rise. Was happening also in the other axis, but was "less destructive". So I have few questions in order to help.

- Could you give a little bit more detail on what happen? 
- Does it always happen at the same place in the same file ?
- What's the state of cncjs when it happen? (Any errors?)
- Could it be happening on other axis or only Z?
- What are you using on the electronic side ? (ex.: CNC Shield, stepper...)
- Do you think it could be a wiring issue(ex.: loose connection) or Interference with unshielded wiring ?
- What lead you to think you lost connection with PC or cncjs issue?

maroussel, 2021-10-06:
Any news on that @thiaggo ?

MitchBradley, 2021-10-06:
Sometimes a failure to change direction can be caused by an electrical timing problem on the DIR signal to the stepper driver.  It can help to add a delay between a change in the DIR signal and step signals.  This has to be done in the controller, not in CNCjs.  I have also seen direction problems resulting from bad connections between the stepper driver and its motor.  In any case, it is extremely unlikely that @maroussel's problem is caused by CNCjs, as opposed to the combination of the controller, drivers and motors.