cncjs/cncjs #677: I can't access the Raspberry Pi and I can't see the video.

asagayakita, 2021-01-07:
#### Description
The first time went well.
When I restarted, the video disappeared.
When I typed the command again, I got an error and it was stuck there. Please help me.

pi@raspberrypi4B:~ $ /usr/local/bin/mjpg_streamer -i "input_uvc.so -r 1280x720 -d /dev/video0 -f 30" -o "output_http.so -p 8080 -w /usr/local/share/mjpg-streamer/www"
MJPG Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: 30
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
libv4l2: error setting pixformat: Device or resource busy
Unable to set format: 1196444237 res: 1280x720
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] RaspberryPi OS
- [ ] Windows
- [ ] Mac
- [ ] Linux

asagayakita, 2021-01-08:
After removing the auto boot when starting Raspberry Pi, the phenomenon disappeared. You can now see the video, if not the ultimate solution.

AustinSaintAubin, 2021-01-20:
Made updates to the [install script for Raspberry PI OS](https://github.com/cncjs/cncjs-pi-raspbian) that should resolve this.

closed 2021-01-20 by AustinSaintAubin