cncjs/cncjs #679: Opening files from Apsire in cncjs

Jozhua61, 2021-02-02:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2021-02-02:
This issue is impossible to understand.  There is no description.  The keyword "Apsire" in the topic is not a word.  If it is a misspelling of "Aspire", a Google search for "Aspire" turns up a definition of the english word, a credit card ad, a retirement planning service, an E-cigarette product, ...

closed 2021-02-02 by MitchBradley

slimline33, 2021-02-02:
just export the file with grbl mm oder grbl inch then open the generated gcode file in cncjs volá