cncjs/cncjs #681: Warn on G28

benlaurie, 2021-02-13:
#### Description

On many machines a G28 is never a good idea - but Fusion 360 and other CAM programs are quite fond of emitting them - would it be possible to get a warning and a chance to cancel (or even ignore)?

#### Versions

- CNCjs: 1.9.22
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

neilferreri, 2021-02-13:
That would be annoying to those correctly using G28.
Why not disable it before you post process? Or use a different post processor?

benlaurie, 2021-02-13:
Sure, I fixed it by removing the option from the post processor, but here's the thing: cncjs gives an XYZ bounding box for the tool, which is very useful for making sure you're not about to do something incredibly stupid (and dangerous). G28 can go far beyond that box and cause serious damage. Sure, perhaps there should also be an "I know what I'm doing and honest G28 is totally fine" checkbox. Or maybe the bounding box should include the G28 - which would obviously require interaction.