cncjs/cncjs #682: Error 11 and 33 with cncjs. NC file works with other softwares.

jonas740, 2021-02-14:
Ive been trying to set my cnc 3018 router up with cncjs since it seems to be awsome but i have ran into problems. i get "error 11" and i have used theese nc files with unversal gcode sender and other kind of grbl software with my router but with cncjs it just wont do it.

Also when pressing "play button" to continue after each error i end up with error 33. The board used is the stock board for my cnc3018. I have also tried a MKS Gen L with grbl on it but that aint working either, it says its doing things but it does not move with that board.

im pasting below from command line what happens and also i have included the nc file im trying to use in .txt format.

CNCjs 1.9.22 [Grbl]
Connected to /dev/ttyUSB0 with a baud rate of 115200
Grbl 1.1h ['$' for help]
client> $$
$0=10 (Step pulse time, microseconds)
$1=254 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=1 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=3 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=0 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=25.000 (Homing locate feed rate, mm/min)
$25=500.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=1.000 (Homing switch pull-off distance, millimeters)
$30=10000 (Maximum spindle speed, RPM)
$31=0 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=800.000 (X-axis travel resolution, step/mm)
$101=800.000 (Y-axis travel resolution, step/mm)
$102=800.000 (Z-axis travel resolution, step/mm)
$110=5000.000 (X-axis maximum rate, mm/min)
$111=5000.000 (Y-axis maximum rate, mm/min)
$112=1500.000 (Z-axis maximum rate, mm/min)
$120=20.000 (X-axis acceleration, mm/sec^2)
$121=20.000 (Y-axis acceleration, mm/sec^2)
$122=20.000 (Z-axis acceleration, mm/sec^2)
$130=200.000 (X-axis maximum travel, millimeters)
$131=200.000 (Y-axis maximum travel, millimeters)
$132=200.000 (Z-axis maximum travel, millimeters)
ok
feeder> G10 L20 P1 Z0
ok
feeder> G10 L20 P1 Y0
ok
feeder> G10 L20 P1 X0
ok
client> ~
> G3 X51.142131979695435 Y-58.647208121827404 I-1.7309644670050761 J-2.200507614213198 F1200 (line=10)
error:11 (Line overflow)
client> ~
> G2 X44.796954314720814 Y-54.22842639593909 I0.5329949238578681 J-0.03553299492385787 (line=21)
error:11 (Line overflow)
client> ~
> G2 X45 Y-53.97969543147208 I1.251269035532995 J-0.8248730964467005 (line=22)
error:33 (Invalid gcode ID:33)
client> ~
> G2 X45.49746192893401 Y-53.54568527918782 I4.378172588832488 J-4.49238578680203 (line=23)
error:33 (Invalid gcode ID:33)
feeder> M5
ok
[fbk_logga_02.txt](https://github.com/cncjs/cncjs/files/5978452/fbk_logga_02.txt)

#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x ] Linux

neilferreri, 2021-02-14:
How are you generating that gcode? Grbl has a limit to the length of a single line sent.

jonas740, 2021-02-14:
> How are you generating that gcode? Grbl has a limit to the length of a single line sent.

This file was made with makercam. Has worked just fine with other grbl software.

What i ended up doing was that i switched back to my Mks gen l board and went through its settings and then cncjs just accepted the cnc code with no complaint.

Only issue was that it routed it mirrored. But I guess that could be the $3=4 setting i currently have.

But with my dedicated grbl board it will not work strange enough.

So the gcode aint the problem it seems.

MitchBradley, 2021-02-15:
Those lines are just crazy long.  For example, look at

G3 X51.142131979695435 Y-58.647208121827404 I-1.7309644670050761 J-2.200507614213198 F1200

Representing the X coordinate to 15 postdecimal digits is silly.  I we assume that the machine is precise to .001 mm - which is extremely precise - then at most we might need 4 postdecimal digits in mm mode, 5 in inch mode.  Any digits beyond that are unnecessary and very wasteful of serial line bandwidth.

This is a know problem with MakerCAM - see https://wiki.shapeoko.com/index.php/MakerCAM#MakerCAM_outputs_too_many_decimals for a workaround

But you really should not use MakerCAM.  It is written in Flash, which is an old technology that is so fundamentally un-secure that modern browsers refuse to support it.  It is an exploit waiting to happen.

closed 2021-02-15 by MitchBradley

jonas740, 2021-02-15:
> Those lines are just crazy long.  For example, look at
>
> G3 X51.142131979695435 Y-58.647208121827404 I-1.7309644670050761 J-2.200507614213198 F1200
>
> Representing the X coordinate to 15 postdecimal [...]

Well my questions wasnt about makercam, it was why cncjs can handle makercam files with mks gen l but not with a dedicated grbl board.

I know the decimals are quite ridiculus and I agree to that. But however it works and has worked great until i stumbled over cncjs, and I just reporting this to help make something great even better.

What cam software do you recommend wich is free and simple?

MitchBradley, 2021-02-15:
CNCjs has not been updated in a long time, so if you expect a fix for this problem anytime soon, you are going to be disappointed. 

It is possible that the two different boards are configured for different line lengths.  It is also possible that some gcode senders transform GCode by, for example, stripping out blanks and perhaps even truncating impossibly-precise numbers.  The range of GCode senders and parsers is almost infinite.  Many are no longer supported, and few of them make enough money to permit full-time, or even part-time development.  Giving definitive answers to questions like this, which involve 4 different components (2 grbl implementations running on different boards, an unsupported CAM system, and CNCjs) is quite difficult and there is little motivation for a developer to spend a lot of time on it.

Free, simple, and good often do not go together.  I use an expensive but very good CAM system - Fusion 360.