cncjs/cncjs #684: Marlin start parsing issue

ejt4x, 2021-02-16:
#### Description

The regex to capture the Marlin start message does not seem to capture the message used by some Marlin builds (latest? not sure if this is a variant or recent change). The code looks for 'start' but I'm seeing 'echo:start'.

proposed updated regex: PR https://github.com/cncjs/cncjs/pull/683

```
2021-02-16T20:19:14.555Z - debug controller:Marlin Connected to serial port "/dev/ttyACM0"     at SerialPort.<anonymous> (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:11532:30)
2021-02-16T20:19:15.629Z - silly controller:Marlin < echo:start
2021-02-16T20:19:15.634Z - silly controller:Marlin < Marlin 509D 2.0.7.2
2021-02-16T20:19:15.640Z - silly controller:Marlin < echo: Last Updated: 2020-10-15 | Author: (V1 Engineering, Ryan, 509D)
2021-02-16T20:19:15.644Z - silly controller:Marlin < echo:Compiled: Jan  5 2021
2021-02-16T20:19:15.645Z - silly controller:Marlin < echo: Free Memory: 3285  PlannerBufferBytes: 1552
2021-02-16T20:19:22.506Z - silly controller:Marlin < DIGIPOTS Loading
2021-02-16T20:19:22.508Z - silly controller:Marlin < DIGIPOTS Loaded
2021-02-16T20:19:22.514Z - silly controller:Marlin < DIGIPOTS Loading
2021-02-16T20:19:22.515Z - silly controller:Marlin < DIGIPOTS Loaded
2021-02-16T20:19:22.518Z - silly controller:Marlin < echo:V82 stored settings retrieved (649 bytes; crc 40447)
```

#### Versions

- CNCjs: 1.9.22
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [X] Marlin 509D 2.0.7.2

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

ejt4x, 2021-02-16:
On further review, this is a bug in recent versions of Marlin: https://github.com/MarlinFirmware/Marlin/issues/20305.

It's not a bad idea to have this workaround in place since I'm sure a lot of people are using affected versions (the fix has not been merged into an official Marlin release yet)

emcniece, 2022-03-26:
👋 Hello! Do you know if this issue persists now that https://github.com/cncjs/cncjs/pull/683 is merged?