cncjs/cncjs #685: Camera not working

tvthiel, 2021-02-18:
#### Description

Hi I installed CNCjs with a Raspberry Pi v2.1 camera, directly connecte to the main board with a flatcable. Everything works except the camera.

The camera connected to the RPi itself works, because:

`vcgencmd get_camera` returns
`supported=1 detected=1
`
and

`raspistill -o testshot.jpg` returns a picture

But CNCjs doesn't "see" the camera whatever I try.

What could be wrong?

Regards,
Tom...

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.22
- Node.js: 10.23.3- 
  npm: '7.5.4',
  http_parser: '2.9.4',
  node: '10.23.3',
  v8: '6.8.275.32-node.59',
  uv: '1.34.2',
  zlib: '1.2.11',
  brotli: '1.0.7',
  ares: '1.15.0',
  modules: '64',
  nghttp2: '1.41.0',
  napi: '7',
  openssl: '1.1.1i',
  icu: '64.2',
  unicode: '12.1',
  cldr: '35.1',
  tz: '2019c' 

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

tvthiel, 2021-02-20:
Is anyone going to help???

HimaniGajjar, 2021-05-12:
Hi,  i am facing same issue.

I installed CNCjs(v1.9.22) on Raspberry Pi 4b, i connected USB camera to the pi.

when i go to turn on webcame in cncjs, camera is not detecting (not showing camera module in webcame setting option)

The camera connected to the RPi itself working fine, i checked with fswebcam and take a click, it returns image.

hope anyone look forward to help, thank you.

Regards,
himani

dzervas, 2021-07-30:
While it's not clear, in order to use the PiCam you're gonna need mjpg-streamer (https://github.com/jacksonliam/mjpg-streamer). This actually makes the picam stream over HTTP to the local network.
Here's how to make it work:

```bash
sudo apt install cmake libjpeg-dev 
git clone https://github.com/jacksonliam/mjpg-streamer
cd mjpg-streamer/mjpg-streamer-expiramental
make
sudo make install
pm2 start $(pwd)/mjpg_streamer -- -o "output_http.so -w $(pwd)/www" -i "input_raspicam.so"
pm2 save
```

Now head over to cncjs and in the settings of the webcam widget put: `http://<raspberryip:8080>/?action=stream`. E.g. my "ip" for the raspberry is `cnc.lan` and I have the following config:
[image]

Keep in mind that the camera feed has to be accessible by the browser, NOT the server in order for it to work. So you can't have mjpg-streamer listening to localhost (much better for security).