cncjs/cncjs #686: Marlin - Rambo 1.4 - jog not working

scfigg, 2021-02-25:
I'm running the latest CNCJS with Rambo 1.4 with the Marlin firmware sent directly from V1. Running CNCJS on a pi. I connect to it fine, but I can not job it with the buttons. I know this was fixed in the release a year ago, but I'm still not able to do it.

Any suggestions? Running on a Pi4 via USB.

scfigg, 2021-02-25:
FYI I just tested it on Windows as well, and no luck.

The output on the console looks weird as well. Not sure if I have something configured incorrectly?

https://imgur.com/a/vN3zSXE

scfigg, 2021-02-26:
Another update - while connected, I can upload GCode via the web ui, but when I hit start, the machine doesn't move either. I can move and print via Repetier or via the LCD.

mickeprag, 2021-02-26:
Looks like you are connecting with the wrong baud rate.

scfigg, 2021-02-26:
I've tried every one of the baud rates, and none of them work once connected.

holterb, 2021-04-01:
make sure that your serial port is correct (depends on the board, but I am using Serial_Port 0 and Serial_port_2 -1 to get connection to CNCJS and the tft screen with a BTT SKR board). CNCjs will say its connecting, even if the port is set incorrect.
Also, when i get garbage on my 3d printer screen it is often caused by interference in the connection from USB to the Marlin Board. I am guessing this is likely your issue, try a shorter cable, remove it from bundles with other wires carrying power or signals, and make sure the USB cord is shielded.

emcniece, 2022-03-26:
Can anyone confirm that this issue is resolved now that https://github.com/cncjs/cncjs/pull/683 is merged? 🙏