cncjs/cncjs #692: Feature/enhancement

SgtKiLLx, 2021-03-28:
idk how active this page is, but im wondering if any further development for cncjs will happen. id like to see the browser UI brought into the 21st century, also maybe the ability to change the theme, or background colour..

MitchBradley, 2021-03-28:
CNCjs development is in a lull at the moment.
Re the UI: ask yourself how you would react if, after having spent thousands of uncompensated hours developing software, someone you don't know leads with. in essence, "your UI is so last century".  Not the best way to establish rapport with a developer.
Re themes: As one of the few people currently trying to support cncjs, themes would make my job harder.  Sometimes the only way to figure out what someone is doing wrong is via a screenshot.  Themes would make "visual pattern matching" harder.
Re background color: A lot of the trivial visuals are controlled by .css files, so in principle you could edit them to adjust simple things like background color - but if you want good support it would be best to revert to the stock setup before asking questions.

SgtKiLLx, 2021-03-28:
Sorry. I'll delete.

closed 2021-03-28 by SgtKiLLx