cncjs/cncjs #693: BIG g-code files

JOKaija, 2021-04-10:
#### Description

Houston, we have a BIG problem... I have a really big routing to do with great accurate.  The problem
is, that it have 5 million lines. YES 5 MILLION! And I need them ALL! I KNOW, IT WILL TAKE DAYS TO MAKE, BUT
I DON'T CARE!!! No, I'm REALLY ARE NOT GOING TO MOVE TOOL PATHS!!! 

So, how about to make a modification, that cncjs reads g code only as a blocks, like 400 000 lines or so, and when
routing done, then it take next 400 000 lines under work...

#### Versions

- CNCjs: 1.9.22

#### Operating System

- [] Not Applicable
- [X] Windows
- [ ] Mac
- [ ] Linux

slimline33, 2021-04-10:
Split the file!

Open with notepad++ and make several files.

JOKaija, 2021-04-15:
And keep awake all night to change a new file when previous routed... No way...
I can code faster my own gcode streamer...

  ----- Original Message ----- 
  From: Christoph 
  To: cncjs/cncjs 
  Cc: JOKaija ; Author 
  Sent: Saturday, April 10, 2021 7:59 PM
  Subject: Re: [cncjs/cncjs] BIG g-code files (#693)

  Split the file!

  Open with notepad++ and make several files.

  —

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus