cncjs/cncjs #695: Compile issues/dependencies

giveen, 2021-04-13:
#### Description

Major Compile Issues

```
npm WARN ERESOLVE overriding peer dependency
npm WARN Found: react@15.6.2
npm WARN node_modules/cncjs/node_modules/react
npm WARN   react@"~15.6.2" from cncjs@1.9.22
npm WARN   node_modules/cncjs
npm WARN     cncjs@"1.9.22" from the root project
npm WARN   29 more (@trendmicro/react-anchor, ...)
npm WARN 
npm WARN Could not resolve dependency:
npm WARN peer react@">= 16.8.0" from react-i18next@10.7.0
npm WARN node_modules/cncjs/node_modules/react-i18next
npm WARN   react-i18next@"~10.7.0" from cncjs@1.9.22
npm WARN   node_modules/cncjs
npm WARN ERESOLVE overriding peer dependency
npm WARN Found: react@15.6.2
npm WARN node_modules/cncjs/node_modules/react
npm WARN   react@"~15.6.2" from cncjs@1.9.22
npm WARN   node_modules/cncjs
npm WARN     cncjs@"1.9.22" from the root project
npm WARN   29 more (@trendmicro/react-anchor, ...)
npm WARN 
npm WARN Could not resolve dependency:
npm WARN peer react@"^16.0.0" from react-datepicker@1.5.0
npm WARN node_modules/cncjs/node_modules/@trendmicro/react-datepicker/node_modules/react-datepicker
npm WARN   react-datepicker@"~1.5.0" from @trendmicro/react-datepicker@1.0.0-alpha.7
npm WARN   node_modules/cncjs/node_modules/@trendmicro/react-datepicker
npm WARN ERESOLVE overriding peer dependency
npm WARN Found: react@15.6.2
npm WARN node_modules/cncjs/node_modules/react
npm WARN   react@"~15.6.2" from cncjs@1.9.22
npm WARN   node_modules/cncjs
npm WARN     cncjs@"1.9.22" from the root project
npm WARN   29 more (@trendmicro/react-anchor, ...)
npm WARN 
npm WARN Could not resolve dependency:
npm WARN peer react@"^16.0.0" from react-datepicker@1.5.0
npm WARN node_modules/cncjs/node_modules/@trendmicro/react-datepicker/node_modules/react-datepicker
npm WARN   react-datepicker@"~1.5.0" from @trendmicro/react-datepicker@1.0.0-alpha.7
npm WARN   node_modules/cncjs/node_modules/@trendmicro/react-datepicker
npm WARN ERESOLVE overriding peer dependency
npm WARN Found: react@15.6.2
npm WARN node_modules/cncjs/node_modules/react
npm WARN   react@"~15.6.2" from cncjs@1.9.22
npm WARN   node_modules/cncjs
npm WARN     cncjs@"1.9.22" from the root project
npm WARN   29 more (@trendmicro/react-anchor, ...)
npm WARN 
npm WARN Could not resolve dependency:
npm WARN peer react@"^16.14.0" from react-dom@16.14.0
npm WARN node_modules/cncjs/node_modules/@trendmicro/react-datepicker/node_modules/react-dom
npm WARN   peer react-dom@"^16.0.0" from react-datepicker@1.5.0
npm WARN   node_modules/cncjs/node_modules/@trendmicro/react-datepicker/node_modules/react-datepicker
npm WARN ERESOLVE overriding peer dependency
npm WARN Found: react@15.6.2
npm WARN node_modules/cncjs/node_modules/react
npm WARN   react@"~15.6.2" from cncjs@1.9.22
npm WARN   node_modules/cncjs
npm WARN     cncjs@"1.9.22" from the root project
npm WARN   29 more (@trendmicro/react-anchor, ...)
npm WARN 
npm WARN Could not resolve dependency:
npm WARN peer react@"^16.14.0" from react-dom@16.14.0
npm WARN node_modules/cncjs/node_modules/@trendmicro/react-datepicker/node_modules/react-dom
npm WARN   peer react-dom@"^16.0.0" from react-datepicker@1.5.0
npm WARN   node_modules/cncjs/node_modules/@trendmicro/react-datepicker/node_modules/react-datepicker
npm WARN deprecated bcrypt-nodejs@0.0.3: bcrypt-nodejs is no longer actively maintained. Please use bcrypt or bcryptjs. See https://github.com/kelektiv/node.bcrypt.js/wiki/bcrypt-vs-brypt.js to learn more about these two options
npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm WARN deprecated mkdirp@0.3.0: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated mkdirp@0.5.1: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated @babel/polyfill@7.4.4:  This package has been deprecated in favor of separate inclusion of a polyfill and regenerator-runtime (when needed). See the @babel/polyfill docs (https://babeljs.io/docs/en/babel-polyfill) for more information.
npm WARN deprecated popper.js@1.16.1: You can find the new Popper v2 at @popperjs/core, this package is dedicated to the legacy v1
npm WARN deprecated core-js@1.2.7: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
npm WARN deprecated core-js@2.6.12: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
```

#### Versions

- CNCjs: 1.9.2
- Node.js: 10.24.1
- NPM: 7.9

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl

#### Hardware

- [x] Raspberry Pi

#### Operating System

- [X ] Linux (DietPi)

orangeworx, 2021-08-13:
Here for the same thing, different versions but still dependency issues.
I suspect this has everything to do with cncjs not having been updated to the new dependencies.
I'm new to cncjs so I'll have to dig into the installation and identify these requirements.

good luck

Pinaute, 2021-10-10:
FROM balenalib/raspberrypi3-node:10

WORKDIR /app
COPY ./ ./

RUN [ "cross-build-start" ]
RUN apt update && apt install -y python3 g++ make
RUN npm install
RUN npm run prepare
RUN [ "cross-build-end" ]

EXPOSE 8000
CMD [ "bin/cncjs" ]

Pinaute, 2021-10-10:
I've got the same issue.

Finally, the compilation from source (v1.9.22) work with docker

On your laptop:
`docker build -t myfork/cncjs:1.9.22 .`
(takes a long time, really)
```
FROM balenalib/raspberrypi3-node:10

WORKDIR /app
COPY ./ ./

RUN [ "cross-build-start" ]
RUN apt update && apt install -y python3 g++ make
RUN npm install
RUN npm run prepare
RUN [ "cross-build-end" ]

EXPOSE 8000
CMD [ "bin/cncjs" ]
```
Move the docker image to the raspberry
`docker save myfork/cncjs | ssh pi@my-ip docker load`

On your raspberry 
`docker run --privileged -p 8000:8000 --detach --name cncjs myfork/cncjs:1.9.22`