cncjs/cncjs #696: Need tooling table for atc.

JOKaija, 2021-04-14:
#### Description

Problem. CNCjs doesn't support ATC. I really would like to continue with cncjs, but there is a 2 things, what it should have:

1) Tool table, where you can save all your tool lengths on store. That when using ie. T2 tool change command with or without H02 (tool register number), it take information for variable for G43 command. T2 only is enough for basic use without H...

2) Tool change by Tx command, M6 is useles if don't have a storage, what can pick up next tool ready for pick up next. So T(tool number) is enough for normal ATC use.

Right now I have modified grbl code in my controller, which have stored tool lengths saved on eeprom by $xxxx registers...
It have also routines to leave and take tool from storage(s). But when I want to start to use bigger storage, eeprom memory can't hold all tool lengths. So, that why I need that tool table option for cncjs.

https://www.youtube.com/playlist?list=PLqeuJKd0xpCmxIIBvFchbCWEU9poV9-T9

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

Pinaute, 2021-09-15:
Hi @JOKaija 
I'm wondering, if you have a tool setter, would you still need this tool table option?

Like this : https://www.youtube.com/watch?v=3ikLcC5NidU

JOKaija, 2021-09-21:
Tool setter? You mean probe.. No, I don't need. I measure my tool length with caliper and
write them to controller settings. When cnc take a next toll, it will update z position automatically
with tool length. So, the tip of tool is always z0, when it touches to working bed...

  ----- Original Message ----- 
  From: Antoine Cavard 
  To: cncjs/cncjs 
  Cc: JOKaija ; Mention 
  Sent: Wednesday, September 15, 2021 11:51 PM
  Subject: Re: [cncjs/cncjs] Need tooling table for atc. (#696)

  Hi @JOKaija
  I'm wondering, if you have a tool setter, would you still need this tool table option?

  Like this : https://www.youtube.com/watch?v=3ikLcC5NidU

Pinaute, 2021-09-22:
Yes probe.

Did you see this comment about the tool library. Does it fit your needs?
https://github.com/cncjs/cncjs/issues/293#issuecomment-379481895
the commit :
https://github.com/ahedderich/cncjs/commit/49566ee6f87f6e417505c2b7502e5b766ccb7b0f

For your tool change application, did you use a single Adruino or an Arduino plus another board dedicated to solenoid valve control?   
Because GRBL firmware seems to take all the memory space and all the pin are used.

JOKaija, 2021-09-23:
I use single arduino. I got probe pin to use for pneumatic valve use. Yep, I just got the whole grbl 1.1h + ATC code to fit the smallest memory size arduino. Woodpecker. 12 bytes left free. If I could use tool table from cncjs, then I could get tool length code & eeprom reservation away. However, woodpecker version was only for demonstration purposes and to proof, that atc will work with it. Normally I use atmega2560 with all my different cnc machines. Plain Cnc, plasma cnc cutter, cnc lathe.

Hanki Outlook for Android<https://aka.ms/AAb9ysg>

#293 (comment)<https://github.com/cncjs/cncjs/issues/293#issuecomment-379481895>

JOKaija, 2021-09-29:
Oh, that modification for cncjs start to be, what I needed. I have to dig it out and examine more...

  ----- Original Message ----- 
  From: Antoine Cavard 
  To: cncjs/cncjs 
  Cc: JOKaija ; Mention 
  Sent: Wednesday, September 22, 2021 8:37 PM
  Subject: Re: [cncjs/cncjs] Need tooling table for atc. (#696)

  Yes probe.

  Did you see this comment about the tool library. Does it fit your needs?
  #293 (comment)
  the commit :
  ***@***.***

  For your tool change application, did you use a single Adruino or an Arduino plus another board dedicated to solenoid valve control?
  Because GRBL firmware seems to take all the memory space and all the pin are used.