cncjs/cncjs #697: New feature suggestion

zmrdko, 2021-04-14:
Hello,
Would it be possible to add feature, which would prevent the XY movement if Z is less than 0 (assuming 0 is top of the stock). I think many users damaged some mills or products when moving the spindle by jogging horizontally and hitting the stock material by accident. Maybe just the popup warning would be great.

MitchBradley, 2021-04-14:
That would break many or most GCode programs

zmrdko, 2021-04-22:
I did not mean this to be active during GCODE program execution. Lets say you start execution of some Gcode - the cutter will get below 0 on Z axis to cut some material (given Z0 is top of the material). Then you hit reset, because you realized you made a mistake in Gcode. You will press go to X0Y0, but since cutter is below top of the material, you will probably break the cutter or damage work piece. This can happen if you forget to move Z up before going to X0Y0. And to prevent from this to happen, there can be just a popup window with warning, if you press X0Y0 while Z  working coordinate is below material (negative Z).

dzervas, 2021-08-02:
You can circumvent this by enabling homing - and requiring homing before movement (both possible in GRBL)

emcniece, 2022-03-26:
It seems like there could be many cases where such a warning would be an annoyance. While crashing bits is certainly a risk, avoiding it is one of the basics of CNC operation.

Is it safe to say that this potential hazard can be avoided by adjusting one's workflow and/or enabling homing? If so, perhaps we can close this ticket.

zmrdko, 2022-03-26:
We sure can!
Thank you for advice on requiring homing

closed 2022-03-27 by emcniece