cncjs/cncjs #702: Incorrect GCode While Probing with Marlin

CorruptName, 2021-05-02:
I am using Marlin on SKR 1.4 Turbo.
When I use the probe widget, It fails because it sends:
```
; Z-Probe
G91
G38.2 Z-20 F10
G90
; Set the active WCS Z0
G10 L20 P1 Z19
; Retract from the touch plate
G91
G0 Z0
G90
```

The problem is:
`G10 L20 P1 Z5`

Since this is not supported by Marlin, I believe it needs to be:
`G92 Z5`
"5" = Touch Plate Thickness

I tried to see where I could change this but I cannot find anything.
Perhaps you should allow this section of GCODE to be user changeable?
That way if someone prefers a different way they can do it.

ledfreak3d, 2021-05-14:
any update on this 
found the same problem with skr pro and marlin

Muralha, 2021-05-21:
Looking at the code, it's hardcoded...
G10 on Marlin = Retract

So, the best solution is to copy the code and make a macro with G92.

Domush, 2022-02-24:
I've added G10 CNC support to Marlin and a few other CNC-specific features, but the Marlin gods don't seem to want to merge them, so I forked it: https://github.com/Domush/mvCNC-Modern-Vintage-CNC-Firmware

It's Marlin bugfix2.0.x, but with CNC features such as both work & machine coord reporting and priority feed rates changes.

emcniece, 2022-03-26:
Sounds like this is outside the scope of CNCjs. Safe to close?

CorruptName, 2022-06-19:
> Sounds like this is outside the scope of CNCjs. Safe to close?

I don't see how this is outside the scope of CNCjs. 
CNCjs should modify this code for use with marlin controllers.

connyg, 2022-10-21:
I  have the same issue, G10 invalid G-code with Marlin firmware.

suprasport, 2024-11-27:
> I've added G10 CNC support to Marlin and a few other CNC-specific features, but the Marlin gods don't seem to want to merge them, so I forked it: https://github.com/Domush/mvCNC-Modern-Vintage-CNC-Firmware
> 
> It's Marlin bugfix2.0.x, but with CNC features such as both work & machine coord reporting and priority feed rates changes.
> 
> My overarching goal of the fork is to create a GRBL-compatible firmware which works seamlessly with any GRBL gCode sender, but with all of the advanced features and diverse hardware support that Marlin provides.

Hi, has anything been sorted with this? I have tried to compile using a skr minie3v3 but it fails on numerous counts, which I presume has something to do with the updates in firmware, having said that it compiles as is without changes