cncjs/cncjs #704: Xpro V5, Grbl, CNCJS

celebie, 2021-06-04:
#### Description

CNCJS "Axes x,y,z button are not activating, buttons are not operating. I used to shuttle my cnc but suddenly it has stopped working. 

#### Versions

- CNCjs: Version 1.9.22 (1.9.22.3668)

#### How Do You Install CNCjs?

- [ ] Download the CNCjs Desktop Application

#### CNC Controller

-  XPRO V5

#### Hardware

- [ ] Desktop or Laptop

#### Operating System

- [ ] Windows

MitchBradley, 2021-06-04:
If the buttons do not activate, it means that CNCjs is failing to communicate with the GRBL controller.  Make sure that you have selected the correct COM port, check the USB cable, and verify that the GRBL controller is getting power. Try talking to the GRBL controller with a different GCode sender application or a serial monitor program to make sure that the GRBL controller is working, the USB cable is okay, and the COM port is correct.

closed 2021-06-04 by MitchBradley

celebie, 2021-06-06:
> If the buttons do not activate, it means that CNCjs is failing to communicate with the GRBL controller. Make sure that you have selected the correct COM port, check the USB cable, and verify that [...]

thanks Mitch,
Firmware Detected: grbl version 1.3a on /dev/tty.usbserial-0001 
I can hear the machine has a connection and responding but I tried with openbuilds controller also axes is not functioning?

celebie, 2021-06-06:
but i think alarm things going on I have no idea what is that for. i also tried unlock the software.

MitchBradley, 2021-06-06:
One thing that can cause an alarm is if the limit switches are tripped at startup - or if they are configured with the wrong inversion state so that they appear to be tripped. https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches .  This is probably a hardware or GRBL misconfiguration, not a CNCjs problem, especially since it does not work with the openbuilds controller.  Please contact the hardware vendor and ask them for help, if you cannot solve it with the information cited above.

karoria, 2021-06-06:
Try "reset" after opening port.

TTALON541, 2022-04-18:
Im having this same issue, my started after I updated to Windows 10 and .24 build.