cncjs/cncjs #706: Server has stopped working -  Is it a true Client/Server architecture?

Bruno-Muller, 2021-06-24:
#### Description

I run cncjs on RaspberriPi 3 B+ and I connect to it using Chrome on an Android tablet.

When a long job is running, the tablet may lock itself or go to sleep mode.

When I unlock the tablet or wake it up from sleep mode, then cncjs shows "Server has stopped working" message. The tty port is closed.

Eventually, when the tablet is in sleep mode for too long, the job may stop in the middle of an operation (like if the next commands have not been queued into the CNC).

With this behavior, I guess that cncjs running on the Rpi is not a standalone server operating the CNC.
It seems that the client web browser is actually acting like the server queuing commands to the CNC.

Is this a bug ? or cncjs has not been designed to be used like that  ?

In addition, I noticed, that cncjs may have not been designed to work with 2 client web browsers at the same time, (I tried once, but I do not do this way anymore)
1 device (desktop computer) to upload the gcode and look at the webcam video stream, and the other device (tablet) to operate the CNC.
The settings are not replicated between client browsers (settings like the tty path and other stuff - I don't remember). It looks they are stored locally on the client web browser and not on the cncjs server.

#### Versions

- CNCjs: 1.9.22
- Node.js: 10.24.0
- NPM: 7.8.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

Fusseldieb, 2021-07-02:
I can't really reproduce your bugs.

cncjs runs on the RPi and I access it using Chrome, I can literally unplug the ethernet cable and it continues running until it's done.
Also I can watch the job from multiple computers just fine.

dzervas, 2021-08-02:
What do the logs of cncjs say?

nickm324, 2021-08-02:
Had similar issues with version .15 and above. If you check the PI you will see the wifi is now inactive and can only be cleared by a reboot, essnetially terminating all access from any remote devices from connecting to the web interface.

Bruno-Muller, 2021-08-03:
Well, I tried a dry run, but of course, it does not happen when I want the issue to occur.... I'll try again later this week.

Please, can you tell me where I can locate the cncjs log file ?

emcniece, 2022-03-26:
I wonder if some of this is caused by Wifi low-power modes: https://www.heelpbook.net/2021/raspberry-pi-4-preventing-wifi-module-to-go-to-sleep-mode/

Log location will depend on how you are running `cncjs`. Your report says you installed with `npm`, are you running the software in terminal with the `cncjs` command? If so, you can add `-v` or `-vvv` to increase log verbosity.