cncjs/cncjs #709: Retrieve the bounding box of the currently loaded GCode

dzervas, 2021-07-30:
I'm working on #32 (more like a modern widget port of https://github.com/kreso-t/cncjs-kt-ext in React) and I'm facing a problem calculating the area that the probing should commence. I currently can't see any easy way to get the bounding box of the currently loaded GCode (`xmin`, `xmax`, `ymin`, `ymax`). They are calculated in https://github.com/cncjs/cncjs/blob/master/src/app/widgets/Visualizer/Visualizer.jsx#L968 but I've got no way of extracting that data from my widget - pubsub used in https://github.com/cncjs/cncjs/blob/master/src/app/widgets/GCode/index.jsx#L218-L247 can't be used by a widget, as the pubsub from what I understand is process-local and can't be accessed by an external widget (so I'd have to integrate my widget in CNCjs, which might not be of your liking).

Is there something I'm missing? Maybe I should open a PR exposing this data about the gcode in the `gcode:load` event?

MitchBradley, 2021-07-30:
Instead of changing the API of the gcode:load event, a better approach would be to create a new event.  Changing existing APIs creates all sorts of compatibility problems with existing code.  Adding a new event provides a much cleaner migration path.

MitchBradley, 2021-07-30:
Another approach would be to grab the bounding box calculation code from cncjs-shopfloor-tablet.  Doing so would prevent having to change cncjs itself.  cncjs code changes are not too difficult, but then if you want to deploy them to others, you run into the release problem.  If you are interested, look at showToolpath in https://github.com/cncjs/cncjs-shopfloor-tablet/blob/master/src/toolpath-displayer.js .  It first processes the gcode to determine the bounding box, then uses the bounding box to compute a scale factor for displaying a 2D projection of the toolpath.  All of the code is in that repo in reasonably plain javascript, in files with names that begin with "simple".

dzervas, 2021-08-01:
It'd be nice to avoid the code complexity - but after all, I'm not sure how many extensions it would favor

dzervas, 2021-08-01:
How would a pubsub to controller event bridge sound? Promotes code reuse and a very easy and eye popping change that has to be done if/when a widget is getting integrated into cncjs with perfect backwards compatibility without braking anything. E.g. pubsub's `gcode:bbox` would become a regular event `pubsub:gcode:bbox`

MitchBradley, 2021-08-01:
The main problem is that nobody is releasing new versions of cncjs.  Coding is relatively easy for people with the right skillset, and a lot of fun.  Release management, maintenance, and support is just work, and a lot of it.

dzervas, 2021-08-01:
Yikes, the ol' maintenance problem... Is the project "halted" in terms of maintenance?

MitchBradley, 2021-08-02:
Look at the commit history...

closed 2021-08-02 by dzervas