cncjs/cncjs #710: No welcome message, no reset, but working with UGS and Lightburn

wildfang, 2021-08-02:
#### Description

I can't establish a connection with Cronos Maker Board using GRBL 1.1h. There is no welcome message and the buttons are grayed out. Reset is not doing anything.

I can,  however, run the machine just fine with UGS and Lightburn, using the same settings.

CNCjs itself is also working fine as it can control my other GRBL controller. It has this issues with the Cronos Maker control board only.

I have limit switches but they are not tripped - disabling them doesn't change anything.

I read somewhere that the Cronos Maker board does not come with a bootloader - is this the problem? Is there a way to get it connect to CNCjs? I get the welcome message with UGS and Lightburn but not with CNCjs.

Any help appreciated, thanks!

#### Versions

- CNCjs: 1.9.22
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM

#### CNC Controller

- [x] Grbl

#### Hardware

- [ x] Desktop or Laptop (Mac Mini Server)

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ x] Mac
- [ ] Linux

[image]

wildfang, 2021-08-02:
Okay, I got it working by wiring a button to the reset header and manually resetting the controller with the button. Yay!

dzervas, 2021-08-02:
I guess you can close the issue now?

wildfang, 2021-08-02:
Yes. Thank you.

closed 2021-08-02 by MitchBradley

noname1020, 2022-04-03:
I used to push reset button in order to connect fluidnc with cncjs,
then I figured that I can give a reset signal by enter $Bye (command for reset MCU in fluidnc equivalent to push reset botton) at cncjs console,so make it easier here is how I did it.
Add an a events at setting like this.
Events - Open a serial port (System only)
Trigger - System
Commands - stty -F /dev/ttyUSB0 115200 cs8 -cstopb -parenb -icanon
sleep 1
echo -e '$Bye' > /dev/ttyUSB0
Save.
Click open botton,no more push physical reset botton.
Hope this helps