cncjs/cncjs #711: [Feature Request] Physical Button Input via RPi GPIO

nelmdog, 2021-08-03:
#### Description

I'm running an MPCNC on an SKR Pro v1.2. As a board designed to control an FDM printer and not a CNC, it has no feed hold header on the board. This omission sent me searching for a way to add this feature, but I have found no obvious implementations to date.

The ability to implement configurable gcode triggers, initiated from local physical buttons connected to the Raspberry Pi's GPIO headers, would greatly extend the safety of Marlin based CNC machines. Further, the general functionality of all RPi driven machines could be improved by allowing specified commands to be immediately and physically available.

[I have forked](https://github.com/nelmdog/button-mod-of-cncjs-pendant-raspi-gpio), poorly, Austin St Aubin's [cncjs-pendant-raspi-gpio](https://github.com/cncjs/cncjs-pendant-raspi-gpio) in an attempt to add this feature, but native support would be infinitely more stable and easier for end-users to implement than my hack-and-slash job.

Thanks for the consideration.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [ ] Download the CNCjs Desktop Application
- [x] V1Pi Preinstalled

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [x] Marlin

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

AustinSaintAubin, 2021-08-08:
See: [cncjs-pendant-raspi-gpio](https://github.com/cncjs/cncjs-pendant-raspi-gpio) as a starting point.

UglyDonkey, 2021-09-01:
Hi
I recently made [cncjs-pendant-gpio](https://github.com/UglyDonkey/cncjs-pendant-gpio). Buttons and gcode are configured in json file which provides much easier way to assign functions than copy and paste code like it is in [this fork](https://github.com/nelmdog/button-mod-of-cncjs-pendant-raspi-gpio) 😄 
I thought about implementing that feature in cncjs but trying to understand what is going on in that code requires more patience and time than I have so I decided to do this as pendant.