cncjs/cncjs #712: error

shaunc123, 2021-08-17:
#### Description
WHEN I TRY TO SET HOME OR ZERO OUT MACHINE POSITION 
I GET THIS ERROR SEE BELOW 

feeder> G28.3 X0
error:20 (Unsupported command)
feeder> G28.2 X0
error:20 (Unsupported command)
>

#### Versions

- CNCjs: 1.9.22

#### How Do You Install CNCjs?

- [ ] Download the CNCjs Desktop Application

#### CNC Controller

 cnc X PRO v%

#### Hardware

- [ ] Desktop o

#### Operating System

- [ ] Windows 10

MitchBradley, 2021-08-17:
GRBL does not support G28.3; GRBL can only home with limit switches.  You can set the work coordinate offset with GRBL, but not the machine zero.  GCode programs operate in work coordinates, not machine coordinates, so the inability to set the machine zero is usually not a problem in practice.

closed 2021-08-17 by MitchBradley

shaunc123, 2021-08-17:
Thank you

Shaun Cronin  ( MANAGING DIRECTOR )
***@***.******@***.***>
www.seceng.com.au<http://www.seceng.com.au>
Phone 02-9524 9952   Mob 0417 257 276
SEC ENG P/L ABN 79 086 478 668
This message may contain confidential and ,or privileged information of SEC ENG P/L
If you are not the intended recipient please
1) do not disclose, copy, distribute or use this information,
2) advise the sender by return e-mail
3) delete all copies from your computer

pentacular, 2021-08-17:
I usually reposition, then close and re-open the device to reset the state and get it to consider the current position as zero.

shaunc123, 2021-08-17:
cheers

Shaun Cronin  ( MANAGING DIRECTOR )
***@***.******@***.***>
www.seceng.com.au<http://www.seceng.com.au>
Phone 02-9524 9952   Mob 0417 257 276
SEC ENG P/L ABN 79 086 478 668
This message may contain confidential and ,or privileged information of SEC ENG P/L
If you are not the intended recipient please
1) do not disclose, copy, distribute or use this information,
2) advise the sender by return e-mail
3) delete all copies from your computer