cncjs/cncjs #715: Simple question: Where are the macros stored ?

BernardG, 2021-09-10:
#### Description
My question is in the subject. I would like to know where are the macros files, as I would prefer to edit them with VS Code, if possible, rather than through CNCjs interface.

#### Versions

- CNCjs: 1.9.22 
- Node.js: 12.22.6
- NPM: 6.14.15

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

neilferreri, 2021-09-10:
You could always copy/paste from your editor of choice.
That said, the macros can be found in the .cncrc file.  I don't have a pi in front of me to see where that is.  On Windows, it's located under C:\Users\"Your User"

BernardG, 2021-09-11:
Thank you. Should be able to find it from there.