cncjs/cncjs #718: [Feature Request] Allow connecting to a remote server through the Electron app

jdforsythe, 2021-10-02:
#### Description

The Electron app should allow entering a remote IP/host to connect to a server.

Use case:

Raspberry Pi Zero WH running cncjs, connected to a 3018pro. UI is flaky on the Pi. Solution would be to run the Electron app on my Mac and interface with the server on the Pi, allowing me to use the app without downloading the assets and have a more reliable connection.

I'd be willing to help with this

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

jdforsythe, 2021-10-03:
Another use case is controlling multiple CNC machines from one computer. So I may set up 2 or 3 machines with attached Raspberry Pis, just running the server, and control them all from one laptop, switching at will

paveltizek, 2021-12-29:
How do you install cncjs on zero (1st gen?). I am trying install on zero 2 bud zero have not enough memory to install.

MitchBradley, 2021-12-29:
The only way that I have found to use cncjs on a pi zero is to run only the server on the pi, running the user interface code on some other machine.

paveltizek, 2021-12-29:
Thank you but how to do it? I tried this several times but every time i ended with error heap out of memory. Last i tried this https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Install-Node.js-via-Package-Manager-*(Recommended)*

MitchBradley, 2021-12-29:
It is not easy.  The best way I have found is to do the build on a PC, copy over a bunch of files that are platform-independent, and then build a single platform-dependent module on the Pi.  https://github.com/cncjs/cncjs/issues/357#issuecomment-489998820

paveltizek, 2021-12-30:
Sounds good but it is not very comfortable. The path I finally took is to install Docker (easy), pull image and install cncjs in it. Next i decide to create own image for easy install to everyone (https://hub.docker.com/repository/docker/paveltizek/cncjs). I will be very happy if someone will try it and write some feedback for it. Thank you for your time.

MitchBradley, 2021-12-30:
I hope you will support it.  I do not have time to support yet another alternative way to install.  There are already about a dozen ways that people have posted in various places, and then disappeared.  There is a CNCjs Users Group on Facebook where new users often show up and ask questions like "I followed the installation instructions and it did not work".  The problem is there are so many different sets of instructions that various people have posted, and they tend to either stop working after a few months when something changes, or only work in very specific situations.

closed 2021-12-30 by MitchBradley

jdforsythe, 2022-01-03:
@paveltizek @MitchBradley 

https://github.com/sprack/docker-raspi-cncjs

The above docker image works great on a pi zero 2. Simply follow the instructions in [this section](https://github.com/sprack/docker-raspi-cncjs#running-cncjs-as-a-service) after you install the Lite version of Pi OS

I had success with v1pi on a pi zero 1, but the UI was slow. It worked mostly fine if you turned off the 3d view. 

---

My hope with this request was that the front and back ends could be easily separated and you could have settings to connect to a remote server, allowing not only running the server-only portion on lesser hardware, but also enabling connecting to multiple CNCs with a single interface.

MitchBradley, 2022-01-03:
@jdforsythe Please join the CNCjs Users Group on Facebook and help people who ask questions about the above.  Most new CNCjs users are on that forum; very few use github.