cncjs/cncjs #721: Has this project been abandoned? Can the maintainers hand off control to active devs?

Domush, 2021-10-17:
#### Description

No commits for over a year. Is this a dead project? It's the best looking gCode sender I've seen, and the only one which supports Marlin outright.

Can this project be revived with some new maintainers?

#### How Do You Install CNCjs?

Download the CNCjs Desktop Application

#### CNC Controller

Marlin

#### Hardware

Desktop or Laptop

#### Operating System

Windows

MitchBradley, 2022-03-22:
Actually @cheton owns that Facebook group too.  He asked me to become co-administrator and I (foolishly?) accepted.  I would be happy to have additional help there, but, sadly, most of the answers that are supplied by other users are some combination of too terse to be useful, misleading, plain wrong, or do not answer the question that was asked.

If some one were to start providing quality answers to questions there, I would gladly hand over the reins - and would also strongly consider giving (or asking Cheton to give) commit permission on this repo.

As a member of the cncjs github org, I have merge permission for PRs, but I hesitate to just merge without careful review, for which I have little time.  If I started to do that it might suck me into more work than I am willing to commit, given the time I already spend with my extensive involvement with the FluidNC project.  If someone else were to start providing insightful reviews of the outstanding PRs, that would be a strong inducement to add them as a member with merge privileges.

But even with merge privileges, pretty much all of the recent PRs are blocked due to Appveyor CI fails.  It is probably an issue with the CI scripts, since they seem to be blocked on the same one.  Fixing that would be another way that someone could prove their ability to be a good maintainer.

emcniece, 2022-03-22:
✋  I'd like to volunteer for fixing the CI, but I'm not sure that it's possible without elevated repo and CI permission. I'll try forking and testing on a separate Appveyor pipeline.

If this can be fixed I would like to volunteer for further CI/delivery, platform compilation, and dependency update efforts. Not sure I can dedicate time to learning the core software at the moment.

Unblocking the PR workflow should be very high priority!

Opening https://github.com/cncjs/cncjs/issues/747

MitchBradley, 2022-03-22:
That is exactly the right approach - fix and test in a fork.  That is a good way to prove one's skills.

balthisar, 2022-03-22:
@emcniece, @MitchBradley, thanks!

@emcniece, let me know how I can support the Appveyor pipeline work. I'll take a look independently, too.

@MitchBradley, I think my intention is to support repairing bugs, and not user technical support, and certainly not on Facebook. I don't envy your having agreed to "foolishly" help out over there! ;-) From the sound of it, it sounds like other users have that aspect covered, so I won't feel guilty ducking out. (Your expectations are completely reasonable, by the way. Thanks for that.)

emcniece, 2022-03-25:
@Domush I've volunteered for a maintainership role for this project and the team has started introducing me to the work. I'm happy to announce that you can look forward to some new releases very soon, perhaps within the next month :)

My focus will be in 2 areas: categorizing and triaging open issues, and documenting/consolidating the CI/CD pipeline so that we can start clearing the factory floor.

I'm not very familiar with the codebase itself and I haven't actually used the software yet, but I'll start playing with it shortly. If there are others interested in helping (@balthisar 😉) an easy way to start might be to sort out stale/outdated issues from active issues. Reproducing and verifying reported issues might help us prioritize high-impact changes. 

I'll leave this issue open for now for @Domush to do with as they see appropriate... and I'll reconsider closing it after we get a new release published 😁

dwery, 2022-03-25:
@emcniece  thank you!

balthisar, 2022-03-26:
@emcniece, I'll start going through the open issues in the next day or two to begin a triage process.

We should settle on some baseline requirements to eliminate issues related to installation environment. With the prevalence of version managers these days, I don't see any harm in choosing a recent LTS version of node as a baseline and ignoring everything else. I'm running CNCjs on node 16 successfully. I've got several physical and virtual environments for testing purposes.

I'm not familiar with Appveyor (yet), but I've had excellent luck doing pure C and some Swift CI using Github native solutions. Github's runners are only x86 currently, though, and I suspect that most folks are using ARM platforms.

emcniece, 2022-03-26:
Gah, the x86 restriction caught me off-guard... [you're right](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners#supported-runners-and-hardware-resources), it seems that arm64 is only available for self-hosted runners. That alone might make it a good idea to stick with Appveyor for now.

I've started labelling issues [in this project](https://github.com/orgs/cncjs/projects/3/views/1) with some relevant categories - feel free to contribute however you see fit.

ScaredyCat, 2024-08-30:
Seems we're back in the same position again.... no commits for nearly a year.

Are there any active forks?

MitchBradley, 2024-08-30:
The only active fork I know of is gSender.