cncjs/cncjs #723: Cannot connect to grbl-esp32 via cncjs on Raspberry pi (RPi)

Di3mex, 2021-11-02:
#### Description

### Install
I have done a fresh install of Raspbian on a Rpi3. Installed nodejs and npm via node version manager. Using node 12 as a default. Installed cncjs in the node 12 install (its in the packages). Pi user has the dialout group and should be able to access devices. I installed the serial port library manually, as that was mentioned somewhere.

### Error
At first I was getting "error opening serial port" in the UI, now it connects but the welcome message (with the settings and version of grbl) of the grbl firmware is not received in the console. The controller also doesn't respond to any commands like reset, unlock, "$", "$$".

### Hardware 
This controller board works though with my Laptop that is running the standalone cncjs electron app (v1.9.22).

I have also added a powered USB hub and measured the voltage on the USB line (5.05V), so I really don't know what else I could try. My Laptop is having performance issues which is leading to the machine pausing during cutting from time to time. So while my laptop setup currently works I need to get a standalone setup going soonish. 

Any help would be greatly appreciated

#### Versions

- CNCjs: 1.9.23
- Node.js: 12.22.7
- NPM: 8.1.2

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2021-11-02:
Try connecting to Grbl_Esp32 with a terminal program like minicom.  That will tell you whether the problem is related to cncjs or just a problem with USB serial in general.

Di3mex, 2022-01-08:
I gave up. Then I reinstalled Raspbian on a spare Pi. It worked on the first try. After rebooting it now it fails to connect. Connecting via minicom gives a garbled response message.

MitchBradley, 2022-01-08:
The fact that it does not work with minicom means that the problem has nothing to do with CNCjs.

closed 2022-01-08 by MitchBradley