cncjs/cncjs #727: Seeking Help Troubleshooting Connection Problems

mschreib-il, 2021-11-24:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [x ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

I'm running cncjs to an XPRO V5 controller which is driving a WorkBee threadscrew 1000mmx1000mm router.  I'm working with 2 different 64bit Windows 10 PCs running on Intel Core iX architectures. 

I can connect successfully from 1 machine but not from the 2nd machine. Both machines find the Xpro V5 com port, which suggests that they can access the controller, but 1 machine connects successfully while the 2nd machine returns an "error trying to connect to com port" message, with no further error messages  anywhere that I can find.   

How can I run cncjs in debug mode in Windows? I checked the .cncrc file but did not see any intuitive way of enabling debugging. Any other troubleshooting suggestions? 

I am able to connect cncjs to the controller from one machine but am unable to connect

MitchBradley, 2021-11-24:
Try connecting from a terminal emulator program like TeraTermPro, PuTTy, or Arduino Serial Monitor.  That will tell you whether the problem is related to CNCjs specifically (unlikely) or is something in the operating system.

I can think of a couple of things that could be causing the symptoms that you describe.  One is an old version of the device driver.  Recent ESP32 modules often use a new variant of the CP2102 USB serial chip that is not supported well by older versions of the CP210x driver.  Another possible problem is that some other program has already connected to the serial port so CNCjs cannot connect to the port that is already in use and locked by the other program.

mschreib-il, 2021-11-25:
The connection problem was resolved by running CNCjs with  Administrator privileges. I am unclear on why this was required on the older machine but not on the newer machine, since both are running Windows10.  Admittedly, I'm not a Windows maven, having spent much more time with Linux and Unix. 

The issue can be closed: I'm still interested in knowing how the Windows/GUI version of CNCjs can be run in debugging mode. I'm  also unclear as to how I would run CNCjs from a PuTTy shell; if someone wants to explain how to do that for future reference that might be helpful.  Also, it seems to me that something like the privilege problem I apparently encountered should have resulted in a log message somewhere, and I found nothing in the session file; is there a log somewhere that I don't know about?

MitchBradley, 2021-11-25:
The requirement to use Administrator mode is very unusual; I do not recall anyone else having that problem.  Therefore, I cannot say why it might have been happening, other than a vague guess of some security attribute that might have been somehow applied to the COM port that connects to the controller.

Re debug mode, it is possible to start the CNCjs app from a command prompt.  Right click its icon and select Open File Location.  In the Explorer window that appears, locate the CNCjs shortcut entry, right-click that and select Properties.  In the pane that appears, Copy all the text that appears in the Target box, being careful to get it all because most of it will be outside the small box.  Open the command terminal window of your choice, e.g. Windows PowerShell or Cmd, and paste the Target text into that window as a command.  After some time, CNCjs will start and the diagnostic output from the CNCjs server will appear in the command window.  If you want to increase the debug level, add -v or -vv to the end of the command.

If there is a log, my brief attempts to find it did not succeed.  The "Electron" system that creates bundled apps is very complicated, on top of the extraordinary complexity of the numerous JavaScripts and the CNCjs code itself.  Finding where it puts stuff is a long investigative process.  The fact that CNCjs runs on so many different platforms in so many different modes makes it impossible for me to keep track of the details of every version.

closed 2021-11-25 by MitchBradley

MitchBradley, 2021-11-25:
Re "running CNCjs from a PuTTy shell": PuTTy is not a shell.  It is a serial terminal program.  You cannot "run CNCjs" from PuTTy, but can use PuTTy to connect directly to the serial (COM) port on the GRBL controller. Doing that will let you test the ability of the computer and its OS and drivers to talk to the controller, taking CNCjs out of the equation.  In that way, one can determine if the problem lies with CNCjs itself, or instead with some other thing that is outside of, and beyond the control of, CNCjs.

The majority of the problems that I field both here and on the Facebook CNCjs User's Group have nothing to do with CNCjs, but rather are misconceptions about how to set up GRBL or other such things, often from people who are barely even aware that GRBL exists and completely unaware of its deep role in the machine's operation.