cncjs/cncjs #730: Upload G-Code button does not work

knipknap, 2021-12-13:
#### Description

I cannot get the "Upload G-Code" button to work. I had previously used CNCjs and it worked fine, but after a reboot I cannot get it to work anymore.
I even installed Raspian fresh from the official image, with a fresh version of CNCjs according to [these instructions](https://cnc.js.org/docs/rpi-setup-guide/) as far as possible (they are outdated; mostly just skipped YARN installation and installed nodejs 14 instead of 7), to no avail, the same result.

In the Chrome debugger I see no errors in the Javascript console, and no attempt of making a network request when I push the button.

CNCjs is correctly connected to my CNC via GRBL, and the control buttons in CNCjs work fine. I just cannot upload any GCode anymore.

#### Versions

- CNCjs: 1.9.23
- Node.js: 14.18.2
- NPM: 8.3.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2021-12-13:
Use these instructions https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Install-Node.js-via-Package-Manager-*(Recommended)*

Nodejs v14 is very unlikely to work with cncjs

knipknap, 2021-12-13:
Thanks for the response!
I just reinstalled based on Nodejs 10 (I didn't see any other differences to the previously used instructions, btw.), but the result is the same, the button still doesn't work.

knipknap, 2021-12-13:
In addition, I now tried the Docker container from here:

https://hub.docker.com/r/davidhamm/cncjs

WIth the same result (this is also running 1.9.23, not 1.9.20 as the docs of the image says).
So I am stumped what might cause this issue.

MitchBradley, 2021-12-13:
Run the cncjs server from the command line with -v and see if any of the logging messages shed a clue.

knipknap, 2021-12-13:
There is no error message. However, I found out that I can add 3D models by dragging them to the canvas. So I guess it's just a UI bug that the button doesn't work.