cncjs/cncjs #732: Cannot create new marcos. Could be centered around save function.

Taruuk, 2021-12-28:
#### Description

Cannot create new marcos. 
Press the the plus sign and I can get to the creation screen but the save never works. 
Only one macro works, its the first one that I created over a year ago. I cant save changes made to this macro either. 

I seem to be Linux handicapped right now cause i can't seem to find NPM. or Node.js versions.....

#### Versions

- CNCjs: 1.9.23 (I first noticed this in 1.9.22)
- Node.js: 6.x(?)
- NPM: 5.x(?)

#### How Do You Install CNCjs?

- [ ] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

plumbjet, 2022-04-12:
Had the same problem, Run as administrator fixed it.

Taruuk, 2022-07-05:
Running as admin did not fix this issue for me. 

Now the whole program refuses to do anything other than open on a non visible window. A different thread all together.

closed 2026-09-21 by cheton