cncjs/cncjs #733: Correct NodeJS version

jangrewe, 2021-12-31:
#### Description

The docs say that NodeJS 8 is unsupported, but the release notes for v1.9.21 say NodeJS 10 should be installed. 
My DietPi/bullseye would have NodeJS 12 out of the box, so now i'm wondering what version is actually the highest supported one for master/latest release?

#### Versions

- CNCjs: 1.9.x
- Node.js: 12.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2021-12-31:
"The docs say"...  - There are many versions of the docs in different places, and even more if you consider various howto's that different people have written.

In my experience, NodeJS 10 is the highest version that I have succeeded with.

jangrewe, 2021-12-31:
Okay, to be precise: the homepage of the official website: https://cnc.js.org/ ;-)

Then i'll give v10 a shot, thanks.

closed 2021-12-31 by jangrewe

MitchBradley, 2021-12-31:
Yeah, the trouble is that the official website has sort of been superseded by the github wiki and changes are not propagated back.  That is what happens when there are not enough donations to make it economically feasible for anyone to do maintenance work. Thousands of users, hardly any contributions.

jangrewe, 2021-12-31:
My virtual 0.02€ donation: forward the domain to the wiki! :-)

MitchBradley, 2021-12-31:
I am not the domain owner.

jangrewe, 2021-12-31:
Neither am i.