cncjs/cncjs #734: Torch height controller THC support

kallaspriit, 2022-01-06:
Hey,

I'm planning a plasma cutter build and CNCjs looks very nice, was planning to use it with CNC xPRO V5 controller.

I'd like to use a torch height controller (such as Proma Compact THC 150 )that automatically changes the Z axis height to be a constant distance away from the material that might be distorting with heat and is thus not flat. Basically it measures the torch voltage and tries to keep it at a certain level by commanding the Z axis to move up or down (there's the arc detected signal, up and down signals controlled by relays).

There's a second type of THC (such as Compact THC SD) that hijacks the STEP and DIR signals when arc is detected but as I understand this only works with external stepper drivers (CNC xPRO V5 has internal ones).

I don't think CNC xPRO V5 has inputs or logic to handle the THC neither internally nor the ability to pass this information (arc, up, down) on to the CNCjs software.

I could image making a simple Arduino sketch that reads the ARC, UP, DOWN signals from the TCH and reports it via serial port. CNCjs could then connect to this serial port to get the required information and when ARC is present, use the UP, DOWN signals to override Z axis control.

1. Has anyone gotten THC to work with CNCjs or has some ideas how to use it externally with controllers that do not support it and have internal drivers such as CNC xPRO V5?
2. Would my THC serial interface idea work and maybe get integrated into the CNCjs software?
3. Should I just look into other software-hardware packages such as LinuxCNC or what would one recommend for a plasma cutter build?

saxa, 2022-01-06:
Hi, I am playing a bit with CNCjs, and also plan to see if I can use it to drive my plasma table with my AC servo drives. But time is a problem, so I am basically stale. Although I have nothing more than a working install right now, I can not help you much.
But I would be very pleased to see how you will solve this THC problem. Since CNCjs is just a kind of sender of Gcode to 
arduino I think it is possible to make it work. Now if this functionality already exists in CNCjs I have no idea. Surely you can make it control the Z axis for the height control. Now the questions is just to discover in which way.

MitchBradley, 2022-01-06:
CNCjs has not had any new feature development for several years so you need to look elsewhere for this feature.

MitchBradley, 2022-01-06:
It needs to be done at a lower level than CNCjs in any case.

closed 2022-01-06 by MitchBradley