cncjs/cncjs #737: Doesn't work select port from JSON in Webui menu v1.9.23

urpylka, 2022-01-21:
#### Description

Since I updated to 1.9.23, doesn't work select port from JSON in webui menu.

I made video where showing how it looks ([link to youtube](https://youtu.be/DORRx7BHxhM)). 

PS: I use [own Dockerfile](https://github.com/urpylka/docker-cncjs/blob/master/Dockerfile)

#### Versions

- CNCjs: 1.9.23
- Node.js: 10.24.1
- NPM: 6.14.12

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

wifiguru10, 2022-02-04:
I can confirm, I also had my "custom com port" option fail when running 1.9.23. Downgrading to v1.9.22 solved the issue.

used: 'sudo npm install -g cncjs@1.9.22'

Using the following config in my .cncrc
{
    "ports": [
        {
            "comName": "/dev/ttyV0",
            "manufacturer": ""
        }
    ],
    "state": {
........

dresco, 2022-05-21:
Likewise, came here to raise a new issue and found this one. 
Can confirm custom com ports are still not appearing in 1.9.25, but are fine in 1.9.22

cheton, 2022-05-22:
Hi @urpylka, @wifiguru10, @dresco 

The `node-serialport` module has renamed `comName` to `path` for the current releases. See https://github.com/serialport/node-serialport/commit/6b85c3d719d610461159721683614905e109744d

Please try using `path` instead of `comName` as below:

```js
"ports": [
  {
  "path": "/dev/ttyV0",
  "manufacturer": ""
  }
]
```

I will update the docs if this works for you.

dresco, 2022-05-22:
> Please try using `path` instead of `comName` as below:

Hi @cheton 

Can confirm that works for me, thanks!

closed 2022-05-25 by cheton