cncjs/cncjs #739: New to CNCjs and having some questions.

Durahl, 2022-01-30:
#### Description

Being a long-term UGC-P user I decided to give CNCjs a try because of the more elaborate macro functionality allowing for a little more work convenience with things like Toolsetter Macros but I'm kinda struggling with a few things I hope are just hard to find and not outright missing:

- The ability to jog to a location from withing the `Visualizer Widget` once a G-Code has been uploaded to it.
UGS-P allows for a `RightClick` > `Jog to cursor location` which, while not _absolutely_ necessary, was still kinda handy.
- The ability to define my own hotkeys.
For controlling my CNC I'm using a Stream Deck ( [short video demonstrating it](https://youtu.be/wk1G4gGxdxE) ) which has Shortcuts applied to the buttons that mirror the shortcuts defined in UGS-P. Like `CTRL`+`SHIFT`+`ALT`+`NumPad7` moving the Toolhead to the center of the Top Left Quadrant of the Work Area ( tiled up in a 3x3 grid ) allowing for a quick move into the general direction instead of painstakingly jogging it there.
I've found both the option to add new `Commands` under the `Axes Widget` > `Settings` and then there's the `Macro Widget` but from where I stand neither of them are ideal for the above given example because a) they do not seem to allow to have their actions bound to a Shortcut ( is there a menu somewhere? ) accessible from my Stream Deck and b) they both unnecessarily clutter up the respective Widget they'd reside in ( I'd ideally prefer Pendant specific Commands to remain hidden ).
There also appear to be `Commands` found under `Settings` that don't appear to be adding clutter to the UI ( which is _potentially_ good ) but at the same time also don't come with a Shortcut feature? TBH, I don't even know how to access them to begin with?!
- It appears that Macros need to be confirmed first before they are being executed which quite frankly, I do not care for a lot especially in anticipation of using them with my Stream Deck - Can this confirmation dialog somehow be supressed?
- Am I right to assume that CNCjs has no continuous/smooth Jogging?

Thanks in advance.

#### Versions

- CNCjs: 1.9.23

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2022-01-30:
CNCjs is not being actively maintained at this time.

fred-c1, 2024-02-18:
@Durahl 
Did you find the answer to your question:
"Am I right to assume that CNCjs has no continuous/smooth Jogging?"
I assume that "continuous/smooth Jogging" means that the motors stop as soon as you release the jog key.
Thanks