cncjs/cncjs #740: Non-standard web socket implementation

dwery, 2022-02-07:
CNCjs uses socket.io to talk between its components and pendants. Unfortunately it's not a standard web socket implementation and thus negates the chance to implement pendants on hw like ESP32, Arduino, etc. It would be nice if a standard library could be used in its place such as https://github.com/websockets/ws or https://github.com/uNetworking/uWebSockets.js, as per https://socket.io/docs/v4/#what-socketio-is-not

urpylka, 2022-02-17:
Nice idea. Until I figured out by using an image with `socat` (I did [one](https://github.com/urpylka/docker-cncjs)) and integrated commands.

```json
{
    "ports": [
        {
            "comName": "/dev/tty.esplink",
            "manufacturer": "ESP LINK"
        }
    ],
    "commands": [
        {
            "title": "Start socat",
            "commands": "/usr/bin/socat pty,link=/dev/tty.esplink,raw,echo=0 tcp:${ESPLINK}"
        },
        {
            "title": "Stop socat",
            "commands": "pkill -f socat"
        },
        {
            "title": "Restart CNCjs",
            "commands": "pkill -f cncjs"
        }
    ]
}
```

Additional information (how to setup ESP8266 and etc) you can find in [my blog](https://urpylka.com/posts/post-35/), but only in russian.

CapnBry, 2023-01-12:
I also use cncjs and a virtual COM port on windows with an esplink as well. Just came looking to see if there was any chance of cutting out the janky COM port middleman to just connect directly to the ESP8266. Hope someday this becomes a reality!