cncjs/cncjs #741: Error response from daemon: manifest for cncjs/cncjs:latest not found: manifest unknown: manifest unknown

luisleonardo, 2022-02-07:
#### Description

I am getting the following error when trying to download the Docker image; "Error response from daemon: manifest for cncjs/cncjs:latest not found: manifest unknown: manifest unknown"
[image]

#### Versions

#### How Do You Install CNCjs?

Via Docker Compose or CLI. Same issue.

MitchBradley, 2022-02-07:
Try to track down the person who published the docker image and ask them.  There are many alternative methods to install CNCjs that various random people have posted.  Those people then disappear, expecting other people to support their installation "improvement".

luisleonardo, 2022-02-07:
I was just following the guide in the wiki of this repository. The other methods can't provide a clean and scalable setup.

Thanks anyway,

emcniece, 2022-03-25:
Confirmed that `docker pull cncjs/cncjs:latest` exhibits this behaviour.

`docker pull cncjs/cncjs:v1.9.22`, the most recent tag, succeeds!

It seems that the `:latest` tag hasn't yet been published: https://hub.docker.com/r/cncjs/cncjs/tags?page=1&name=latest

Labelling for followup.

AlTaiRius, 2022-10-04:
:latest tag still not published. Would be good to get this basic docker functionality sorted.

emcniece, 2022-10-04:
Yeah, agreed. I'll dig around for the source CI again, think I found it a few months back.

closed 2022-10-05 by emcniece

emcniece, 2022-10-05:
Latest tag should now be available!

```
$ docker pull cncjs/cncjs
Using default tag: latest
latest: Pulling from cncjs/cncjs
f606d8928ed3: Pull complete
1f34ee6937ae: Pull complete
f2c875846028: Pull complete
60fef9cf4d89: Pull complete
cdfbb6dc4fca: Pull complete
Digest: sha256:304a41f8f178a08ee4aad7cd8a55399a38b038a8ab702e61949b20f57193b003
Status: Downloaded newer image for cncjs/cncjs:latest
docker.io/cncjs/cncjs:latest
```

AlTaiRius, 2022-10-05:
Awesome...thanks for the prompt fix!