cncjs/cncjs #742: "sign in" required for cncjs

NanookMacy, 2022-02-18:
#### Description
I'm trying to use cncjs, but when I open the app I'm prompted to sign in. I haven't used it in 3 weeks due to waiting for a new stepper driver. In the FAQ it directs you to look in the ~/.cncrc file, but I haven't a glue how to do this. Can anyone help or know of a work around.

Thanks

emcniece, 2022-03-31:
How have you installed CNCjs - is this the downloaded app version?

If yes, does the problem resolve if you uninstall and reinstall?

emcniece, 2022-03-31:
Also, what operating system are you using?

This comment makes note of the `.cncrc` file location: https://github.com/cncjs/cncjs/issues/715#issuecomment-917016047

> On Windows, it's located under C:\Users\"Your User"

neilferreri, 2022-04-01:
If you forgot your password, find user account settings in the ~/.cncrc file, and remove all existing users:

Linux / Mac: `~/.cncrc` (e.g. /home/pi/.cncrc)
Windows: `%USERPROFILE%\.cncrc` (e.g. C:\Users\UserName\.cncrc)
`"users": []`

This is taken directly from the FAQs here.
[https://cnc.js.org/docs/faq/#forgot-your-password](https://cnc.js.org/docs/faq/#forgot-your-password)

Open the .cncrc file with your text editor, remove the users, and save the file.

NanookMacy, 2022-04-01:
Thanks for the replies but I've deleted cncjs and bought Mach 3

emcniece, 2022-04-01:
Thanks for following up! Closing this ticket as the answer is provided in thread and in documentation.

closed 2022-04-01 by emcniece

neilferreri, 2022-10-11:
> This comment makes note of the .cncrc file location: #715 (comment)

- Linux / Mac: ~/.cncrc (e.g. /home/pi/.cncrc)
   - Windows: C:\Users\*User_Name*\.cncrc)

Open the .cncrc file in a text editor and remove all users.