cncjs/cncjs #744: Update the Raspberry Pi Setup Guide

mathojojo, 2022-03-13:
#### Description

Hello,

I have a RPi 4, running 64bit raspberry OS version, everything is up to date.
I tried twice to install cncjs, using the Package Manager method, and NVM method, but both failed. Installing the arm64 deb file was not better.
Actually I have cncjs online in pm2 list, but can't connect to it (Rpi_Ip:8000 -> ERR_CONNECTION_REFUSED)

I had to install npm (apt install npm), it's not said in the guide
I got many warnings when installing cncjs with npm. And iptables commands are not availbable, because not installed (nftables is default now on Bulleyes).
I installed iptables, and could use : 
sudo update-alternatives --set iptables /usr/sbin/iptables-nft
sudo update-alternatives --set ip6tables /usr/sbin/ip6tables-nft
sudo update-alternatives --set arptables /usr/sbin/arptables-nft
sudo update-alternatives --set ebtables /usr/sbin/ebtables-nft

Then I could finally use : sudo iptables -t nat -I PREROUTING -p tcp --dport 80 -j REDIRECT --to-port 8000
 and   sudo iptables -t nat -I PREROUTING -p tcp --dport 80 -j REDIRECT --to-port 8000

I rebooted, but still can't access cncjs on a browser....

I think that someone should update this guide for recent Rpi setup.

Thank you

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2022-03-13:
It would be nice is "someone" were maintaining cncjs in general.  But it would be a lot of work and there is not much incentive.

mathojojo, 2022-03-13:
Ohh, do you mean that cncjs is a "dead" project ?

MitchBradley, 2022-03-13:
Look at the commit history

MitchBradley, 2022-03-13:
I have been trying to support it on the Facebook CNCjs User's Group, but I am pretty tired of doing so.  Lots of people appear to be using CNCjs, but few appear to have donated anything so there is no incentive for anyone to maintain the code.  The primary developer has pretty much dropped out.  I have the skills to maintain it but not the desire.  I am spending all of my development time on the FluidNC project.

mathojojo, 2022-03-13:
Yes I noticed. So, it will still work as long as dependencies are available...
Really bad news, I was really enjoying to replace my Web GRBL with CNCjs.
I understand that you don't want to put more effort. Donation are more easy to get with growing projects with a lot of expectations, than with declining projects.

So, it's maybe not the place for that, but would you advise me bcnc instead, or something else if I can't get cncjs to work ?

dwery, 2022-03-13:
cncjswworkanmost fine, we would just need someone who could merge PRs

MitchBradley, 2022-03-13:
The setup guide in the wiki was written and revised by many different people at different times, and tends to break every few months when different versions of this, that, and something else come out, so it is very close to unmaintainable.  That problem is endemic to the Raspberry Pi ecosystem and Linux in general.  That is one reason why there are, to first approximation, no viable commercial applications for Linux.  Anything that you think you know about administering a Linux system will be wrong soon, or is already wrong for most other versions.

The version of the setup guide that I am responsible for, and the one that was most recently maintained, is the "Package Manager" version.  Too bad that it doesn't work.  I don't have a Pi4 and am not going to buy one just so I can fix the wiki, and I am especially not going to re-test everything every time a new version of Raspbian or whatever comes out.

I don't think the iptables thing is actually necessary.  I just start cncjs on port 8000 and access it from there, by adding :8000 to the host IP address in the browser.  localhost:8000 on the same machine, e.g. 192.168.0.100:8000 across the LAN.

You might be able to start it on port 80 by putting `--port 80` on the cncjs command line.

I don't want to recommend a particular other UI agent; there are many of them with different strengths and weaknesses.  I have no idea what matters most to you.  I personally use WebUI with FluidNC because that is what I am developing for.

MitchBradley, 2022-03-13:
> cncjs works almost fine, we would just need someone who could merge PRs

That's easy - just fork it and then you can merge PRs - and then you can respond to issues and fix bugs.

dwery, 2022-03-14:
> > cncjs works almost fine, we would just need someone who could merge PRs
> 
> That's easy - just fork it and then you can  be merge PRs - and then you can respond to issues and fix bugs.

It’s hard for a fork to get traction. Would be much easier if the original dev could just endorse someone.

phalix, 2022-04-09:
Hi, couldnt it be an option to just provide the version of cncjs for what the guide is working for. the current version needs, at least I think, Node JS Version 8, due to the required serialport version, which uses the ... operator.

MitchBradley, 2022-04-10:
> Hi, couldnt it be an option to just provide the version of cncjs for what the guide is working for. the current version needs, at least I think, Node JS Version 8, due to the required serialport version, which uses the ... operator.

Many things are possible if someone takes the responsibility for doing the work, keeping it up to date and answering user questions when something goes wrong.  Are you volunteering?

phalix, 2022-04-10:
@MitchBradley Hi Mitch, thanks for your invite for contributing. If you do not mind, I could test to write some docker images, for various versions. in that way we could always check if installation is still valid. Based on this, I could update the guide.

MitchBradley, 2022-04-10:
Any additional testing and testing infrastructure is always helpful.  To solve the problem, it needs to be backed up with periodic effort to track new versions of everything, especially Raspbian releases,  One of the most troublesome problems, based on some years of trying help new users on the Facebook CNCjs Users Group, is the proliferation of different versions of Raspberry Pi setup instructions in different places.  Typically, a user will find some random version of the instructions, follow that, and it will fail because it is horribly (as opposed to just a little) out of date.  What is particularly troublesome is disk images, which are insidious because they are easy to install and get working at the basic level, making the user think they are home free.  But then they run into a bug in the version of cncjs or a pendant that is baked into the disk image, one that was long since fixed, and they are just stuck.  And the developers are really irritated because the stupid bug that they fixed years ago just keeps coming back to haunt them.  User-facing docker images are not quite as bad as disk images because they are easier to keep up to date, but they can be troublesome for new users because now, in addition to all of the complexity of CNCjs and CNC in general, there is another big component to learn to administer - Docker itself.  Programmers have already bitten the Docker bullet because they need it for other reasons, but for most new users, it is just another "whaaaaat?" component on top of a pile of bewilderment.

phalix, 2022-04-10:
That is not what I meant quite exactly. I actually meant to use docker images in order to see if the current release still compiles against the supported node versions. The current release does not support any nodejs version prior 8. Nevertheless the instructions tell you to use version 4 (with nvm) or 7 with apt-get. This is not working. Also updating the npm to the latest version is not working. Since it will not work with a version quite as old as 8. The current version of process manager is also not working. 

So basically for the current release 1.9.24, we need to have nodejs 8.17.* and pm2 at version 4.5.6. I will try to bring a pull request these days for the documentation in order to close this issue.

MitchBradley, 2022-04-10:
"The instructions": which ones exactly?  "The instructions" implies that there is one set of instructions that everybody agrees on, but in fact there are at least 6 different sets of instructions in this repo's wiki - 3 in the Raspberry Pi Setup Guide section and 3 in the Installation section.  None of them refer to version 4 or version 7.  The references to versions 4 and 7 are in the ancient Raspberry Pi Setup Guide at [cncjs.org.](https://cnc.js.org/docs/rpi-setup-guide/) .

So my basic point is that there are too many conflicting sets of instructions.  The solution is not to fix individual sets piecemeal, but rather to eliminate all but one set, and maintain that set.

MitchBradley, 2022-04-10:
Apparently I already tried to fix [the instructions at cncjs.org](https://cnc.js.org/docs/rpi-setup-guide/) .  The backing store for those instructions is at https://github.com/cncjs/cncjs.org/blob/master/pages/docs/rpi-setup-guide/index.md .  In Feb 2021 I pushed commit https://github.com/cncjs/cncjs.org/commit/e42350b which replaces all of the outdated instructions with a pointer to the [*Recommended* instructions in this wiki](https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Install-Node.js-via-Package-Manager-*(Recommended)*). The rendered view of the updated page looks like:

[image]

But apparently simply editing that page did not make the edits go "live", because the old stale instructions still appear if you browse directly to the cncjs.org link.

emcniece, 2022-04-10:
Does the [deploy.sh](https://github.com/cncjs/cncjs.org/blob/master/scripts/deploy.sh) script need to be run?

MitchBradley, 2022-04-10:
> Does the [deploy.sh](https://github.com/cncjs/cncjs.org/blob/master/scripts/deploy.sh) script need to be run?

Probably.  The last deployment was in 2017.

It appears that the script can be run either from Travis or from a local environment.  I don't know how to trigger the former or what setup is necessary for the latter.  What little Travis knowledge I ever had has long since evaporated.

emcniece, 2022-04-10:
No worries! Travis CI should be replaced with Github Actions, I'll add this to the list 👍

phalix, 2022-04-10:
> Apparently I already tried to fix [the instructions at cncjs.org](https://cnc.js.org/docs/rpi-setup-guide/) . The backing store for those instructions is at https://github.com/cncjs/cncjs.org/blob/master/pages/docs/rpi-setup-guide/index.md . In Feb 2021 I pushed commit [cncjs/cncjs.org@e42350b](https://github.com/cncjs/cncjs.org/commit/e42350b) which replaces all of the outdated instructions with a pointer to the [_Recommended_ instructions in this wiki](https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Install-Node.js-via-Package-Manager-*(Recommended)*). The rendered view of the updated page looks like:
> 
> [image]
> 
> But apparently simply editing that page did not make the edits go "live", because the old stale instructions still appear if you browse directly to the cncjs.org link.

Never found this page, before installing cncjs. And I think it is great guide for people wanting to start using cncjs. Nevertheless, does it not make sense to mention the particular version of cncjs for this guide. As soon as somebody updates the sources, which eventually would require a newer version than 10, the guide would be outdated. If the version is mentioned and the release is not changed, the guide will always be true. And it keeps people frustrated, because of failing installations. I was mentioning before the website cnc.js.org, which as you mentioned has a nowadays faulty instructions. In addition I think it makes sense to update the package.json, since the current release does not work with > 6. It needs >8.  I am happy to support/help (as you were asking earlier) but of course this would require me to do pull requests.

MitchBradley, 2022-04-10:
> No worries! Travis CI should be replaced with Github Actions, I'll add this to the list 👍

Great!  I do know how to use GitHub actions - at least until the next thing comes along and displaces it.

MitchBradley, 2022-04-10:
> Nevertheless, does it not make sense to mention the particular version of cncjs for this guide. As soon as somebody updates the sources, which eventually would require a newer version than 10, the [...]

Take your pick of maintenance tasks - mention the cncjs version or mention the node version.  Either way you have to update the guide periodically.  Furthermore, over time, both have to be updated, because Node versions become too stale.

If you want to take over maintenance of that page I would be thrilled.  I am really burned out.

LongLiveCHIEF, 2022-05-28:
Right now the biggest problem with getting this to work on the most recent rpi and debian builds is that python2 is no longer installed at all on the most recent distros.  This is a problem because running `npm install` will try to do a node-gyp step for serialport bindings during the post-install, and that particular package of serialport that it is trying to build requires python2 to in order build.  

Serialport has recently moved @serialport/bindings to a different package, and so there's dependency chain issues as well even if this does get fixed upstream.

So, building with node 10 or 12 would be my recommendation, but you'd want to be building from buster family or older on debian based distros. (ubuntu 18, debian 10)

LongLiveCHIEF, 2022-05-28:
Additionally, the docker image isn't multi-arch, which means it won't install on raspberry pi based images.  It also can't be built locally, because the docker image doesn't run npm build, it requires that the npm build be run on the host machine, and then the docker build can be run, so you run into the serialport bindings issue all over again.