cncjs/cncjs #745: Grbl 1.3a ['$' for help] [MSG:Check door] >

CNCENTERO, 2022-03-17:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MOTORS NOT MOVE
CNCJS SHOW THIS...
Grbl 1.3a ['$' for help]
[MSG:Check door]
>CAN ANYBODY HELP ME???THANKS

MitchBradley, 2022-03-17:
This is not a CNCjs problem, but rather an issue with the way that your Grbl controller is configured.  It is expecting a safety door switch to be connected to some pin on your controller.  Since nothing is connected, the Grbl controller thinks that the safety door is open and refuses to operate.

Instead of asking this question on a CNCjs forum, please ask the vendor of your controller for help.  You paid money for their product and they should provide some support in exchange.  The CNCjs people have not received anything of value from either you or from the controller vendor.

closed 2022-03-17 by MitchBradley

CNCENTERO, 2022-03-17:
GRACIAS SEGUIRE INVESTIGANDO

CNCENTERO, 2022-03-17:
MI CONTROLADOR ES XPRO V5 CNC