cncjs/cncjs #746: CNCjs stops during job without error

fragsalat, 2022-03-18:
#### Description

Hey there, I experience the issue of CNCjs stopping randomly during a job run. I googled around a bit and couldn't find a solution yet, which is why I ask here now. It seems like the job just doesn't continue at some point without any error message even though I run with -vvv. Here is the log where the execution just stops.

```
2022-03-18T11:00:53.219Z - ESC[35msillyESC[39m controller:Grbl < ok
2022-03-18T11:00:53.220Z - ESC[35msillyESC[39m controller:Grbl > X175.597 Y91.984
2022-03-18T11:00:53.391Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.559,-51.333,1.050,WPos:191.998,39.211,-1.000>
2022-03-18T11:00:53.620Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.560,-49.466,1.050,WPos:191.999,41.079,-1.000>
2022-03-18T11:00:53.873Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.560,-47.587,1.050,WPos:191.999,42.958,-1.000>
2022-03-18T11:00:54.116Z - ESC[35msillyESC[39m controller:Grbl < ok
2022-03-18T11:00:54.118Z - ESC[35msillyESC[39m controller:Grbl > X174.64 Y91.988
2022-03-18T11:00:54.124Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-45.708,1.050,WPos:192.001,44.836,-1.000>
2022-03-18T11:00:54.390Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-43.715,1.050,WPos:192.001,46.829,-1.000>
2022-03-18T11:00:54.640Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-41.841,1.050,WPos:192.001,48.703,-1.000>
2022-03-18T11:00:54.889Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-39.969,1.050,WPos:192.001,50.576,-1.000>
2022-03-18T11:00:55.140Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-38.100,1.050,WPos:192.001,52.444,-1.000>
2022-03-18T11:00:55.417Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-36.036,1.050,WPos:192.001,54.508,-1.000>
2022-03-18T11:00:55.668Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-34.164,1.050,WPos:192.001,56.381,-1.000>
2022-03-18T11:00:55.906Z - ESC[35msillyESC[39m controller:Grbl < ok
2022-03-18T11:00:55.908Z - ESC[35msillyESC[39m controller:Grbl > X172.724 Y91.994
2022-03-18T11:00:55.922Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-32.272,1.050,WPos:192.001,58.272,-1.000>
2022-03-18T11:00:56.172Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-30.403,1.050,WPos:192.001,60.141,-1.000>
2022-03-18T11:00:56.441Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-28.388,1.050,WPos:192.001,62.156,-1.000>
2022-03-18T11:00:56.692Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-26.506,1.050,WPos:192.001,64.039,-1.000>
2022-03-18T11:00:56.941Z - ESC[35msillyESC[39m controller:Grbl < <Run,MPos:3.561,-24.643,1.050,WPos:192.001,65.901,-1.000>
2022-03-18T11:01:02.250Z - ESC[34mdebugESC[39m controller:Grbl Continue status report query: timespan=5067ms     at queryStatusReport (/home/pi/.nvm/versions/node/v10.24.1/lib/node_modules/cncjs/dist/cncjs/server/controllers/Grbl/GrblController.js:711:15)
2022-03-18T11:01:07.307Z - ESC[34mdebugESC[39m controller:Grbl Continue status report query: timespan=5057ms     at queryStatusReport (/home/pi/.nvm/versions/node/v10.24.1/lib/node_modules/cncjs/dist/cncjs/server/controllers/Grbl/GrblController.js:711:15)
2022-03-18T11:01:12.353Z - ESC[34mdebugESC[39m controller:Grbl Continue status report query: timespan=5047ms     at queryStatusReport (/home/pi/.nvm/versions/node/v10.24.1/lib/node_modules/cncjs/dist/cncjs/server/controllers/Grbl/GrblController.js:711:15)
2022-03-18T11:01:17.393Z - ESC[34mdebugESC[39m controller:Grbl Continue status report query: timespan=5039ms     at queryStatusReport (/home/pi/.nvm/versions/node/v10.24.1/lib/node_modules/cncjs/dist/cncjs/server/controllers/Grbl/GrblController.js:711:15)
2022-03-18T11:01:22.440Z - ESC[34mdebugESC[39m controller:Grbl Continue status report query: timespan=5048ms     at queryStatusReport (/home/pi/.nvm/versions/node/v10.24.1/lib/node_modules/cncjs/dist/cncjs/server/controllers/Grbl/GrblController.js:711:15)
```
From there it just continues with the status report query but doesn't continue.

CNCJs is running on a raspberry pi 3b and connected to an Arduino with newest GRBL  (cloned the repo before 1 month). The GCode is generated by Fusion 360 using relative coordinates. 

To me it seems like the arduino somehow crashed but Idk how to debug this. Even though I disconnect and connect to the serial there comes no answer anymore when typing $$. I have to restart the arduino to get it working again.
```
2022-03-18T11:35:00.731Z - ESC[34mdebugESC[39m controller:Grbl Connected to serial port "/dev/ttyACM0"     at SerialPort.<anonymous> (/home/pi/.nvm/versions/node/v10.24.1/lib/node_modules/cncjs/dist/cncjs/server/controllers/Grbl/GrblController.js:1053:13)
2022-03-18T11:35:11.017Z - ESC[34mdebugESC[39m service:cncengine socket.write("/dev/ttyACM0", "$$^M", {"__sender__":"7201cdb9-62bf-4d47-9cbb-46b5efc8fc1d"}): id=gT0bKbjyD9B7JncVAAAn     at Socket.<anonymous> (/home/pi/.nvm/versions/node/v10.24.1/lib/node_modules/cncjs/dist/cncjs/server/services/cncengine/CNCEngine.js:435:15)
2022-03-18T11:35:11.019Z - ESC[35msillyESC[39m controller:Grbl > $$
```

Here are my grbl settings
```
> $$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=801.457 (x, step/mm)
$101=801.000 (y, step/mm)
$102=706.541 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
ok
```

As you can see the hard limits are off so a flickering signal here should not be the problem. Also from the step rate I should be below the 30kHz what was mentioned here. I'm also using a shielded USB cable. Do you have any idea what this could be or how I can debug the GRBL to get the error out there? 

#### Versions

- CNCjs: 1.9.23
- Node.js: 10.24.1
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

emcniece, 2022-03-25:
Hey, I'm new to this project but I have some thoughts: I wonder if motor noise is coupling back and disrupting comms. Does the program stop at the same instruction, or after the same time interval each time? Is there a way to reproduce the serial port freezing without connecting the Arduino to motor drivers?

I'd guess normal programs won't run without position feedback, so maybe there's a gcode instruction set that could execute forever without needing position feedback, like a home operation or motor/spindle speed change.

darkman100, 2022-04-09:
Hey, 

i can confirm the same issue. cncjs stops at a random point during the execution of any program. It could happen after a few seconds, minutes, or never. 

I have a RaspiPi and GRPL (https://github.com/MakeMagazinDE/MaXYposi_Grbl_644) running. Please see the link for more detailed information about the config. I tried both connections from the raspi to the micro controller. The pure connection via the pinout over a TTL converter or with a FT232 programmer which uses the USB device.

I believe the data connection will break down or a bit flipping happens during the execution of the program. Sometimes there are elements like 
`o*
ok▯
o▯`
shown at the console log. 

I have already tried to find the ok parser statement at the code but I haven't found it. If the parser will ignore such wrong statements and the program didn't stop the issue is found but of course not solved, yet.

slimline33, 2022-04-09:
My setup is a Raspberry 3B and just cncjs on it. Nothing happend. Everything runs fine.

When your grbl console shows some unexpected signs then the problem is the usb connection between your raspberry and your controller!

fragsalat, 2022-04-10:
Is there an alternative to the usb connection which seem to be error prone? I already have a shielded cable and try to put it away from electrical noise like the motors.

slimline33, 2022-04-10:
Hi Thomas, mein Kabel vom raspberry zum Board geht über den seriellen Ausgang und hat ca 15cm. Ich habe als Verbindung ein cat7 Kabel verwendet.

Ich hatte die Probleme auch und sämtliche USB Kabel versucht.

MitchBradley, 2022-04-10:
> Is there an alternative to the usb connection which seem to be error prone? I already have a shielded cable and try to put it away from electrical noise like the motors.

Most 8-bit GRBL controller boards support only some form of serial, either direct serial or USB.  There are some controllers that support Ethernet, WiFi, or BlueTooth.  CNCjs only supports serial - direct or USB.  There are third-party programs that will bridge from serial to other forms of networks - CNCjs sees a serial port but the data is transferred over a different kind of network interface.  That would only work if you have a controller with a different interface.

You can also buy isolated USB interfaces that do not require a common ground connection between the controller and the CNCjs machine.  That could reduce interference resulting from currents traveling over the ground connection.

Spindles are typically worse sources of interference than stepper motors.  Sometimes it helps to add power line filtering to the spindle supply.

fragsalat, 2022-04-10:
> Hi Thomas, mein Kabel vom raspberry zum Board geht über den seriellen Ausgang und hat ca 15cm. Ich habe als Verbindung ein cat7 Kabel verwendet.
> 
> Ich hatte die Probleme auch und sämtliche USB Kabel versucht.

Cool. Kannst du mir sagen welche Pins man dafür verbinden muss? Cat7 hab ich gewiss noch rumliegen.

> Most 8-bit GRBL controller boards support only some form of serial, either direct serial or USB. There are some controllers that support Ethernet, WiFi, or BlueTooth. CNCjs only supports serial - [...]

Thanks for the clarification. I wonder why not so many people have this issue. It seem to be quite reproducable. I'll try the recommendation from slimline33 to use a cat7 cable for serial connection to get a harder shielded cable.

slimline33, 2022-04-10:
Es stimmt schon was mitch sagt. Die serielle Verbindung geht nur wenn dein cnc Controller das Auch unterstützt.

https://www.abelectronics.co.uk/kb/article/1035/serial-port-setup-in-raspberry-pi-os

Rx kommt am tx und tx kommt am Rx. Ich weiß nicht mehr ob ich in der boot config etwas geändert hatte aber es war sehr einfach. Ich habe die 5v vom Kabel nicht verbunden. Nur Rx tx über gnd!

MitchBradley, 2022-04-10:
> I wonder why not so many people have this issue.

Actually, it is quite a common problem.  The low-cost off-the-shelf hardware that is common in the DIY CNC world tends to have USB interfaces because they are inexpensive at both the controller and computer end.  Anything that you do to fix the problem in the hardware adds cost, and thus reduces the market penetration of the solution.  Since people tend to buy the cheapest hardware, it is guaranteed that the problem will be widespread.

freexmonster, 2022-04-14:
I also encountered the same problem, usually cncjs orangpi zero usb, I tried to connect to arduino and mks dlc32 and everywhere I get a spontaneous stop during operation. I already have hair on end.

freexmonster, 2022-04-14:
upd:  yes i say dmesg

 usb usb6-port1: disabled by hub (EMI?), re-enabling...

EMI - Electromagnetic Impulse   

[ 2359.530758] usb usb6-port1: disabled by hub (EMI?), re-enabling...
[ 2359.537040] usb 6-1: USB disconnect, device number 6
[ 2359.537670] usb 6-1: ch341_read_int_callback - usb_submit_urb failed: -19
[ 2359.549321] usb 6-1: failed to send control message: -19
[ 2359.555190] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 2359.555301] ch341 6-1:1.0: device disconnected

MitchBradley, 2022-04-14:
CNC machines can generate a lot of electrical noise, and USB is quite susceptible to such noise.  Spindles are particularly bad sources of electrical noise.  Sometimes you can solve it by putting the spindle on a different circuit, or adding a power line filter module to the spindle, or using a shielded USB cable, or adding a galvanic isolator module in the USB circuit, or improving the system grounding.  Search the web for CNC EMI.  It is a complicated topic that must be solved outside of CNCjs.

closed 2024-10-18 by cheton