cncjs/cncjs #747: Appveyor CI not reporting completion status for PRs

emcniece, 2022-03-22:
#### Description

Several open pull requests are unable to be merged as the Appveyor CI is not reporting completion status for one of its checks. 

Example: https://github.com/cncjs/cncjs/pull/708

[image]

Requires CI troubleshooting: clone repo, set up a separate CI pipeline, and test.

emcniece, 2022-03-22:
## Notes

- Platform builds take a long time (over 1h). Devs should get testing feedback faster on their work. Do we need to build 3xLinux + Mac + 2xWin for every commit on every PR?
- A [similar report](https://help.appveyor.com/discussions/problems/18132-build-did-not-start-with-pr) mentions a missing webhook. How was Appveyor added to this repo? Is it missing a [pull_request webhook](https://github.com/cncjs/cncjs/settings/hooks)?
- *There is a plaintext secure token in appveyor.yaml*

While probing the secure token for permissions and exploitability, Github API responses were 401 Unauthorized. This token probably worked at some point, and perhaps since it is improperly stored in plaintext VCS it was either abused or revoked by Github itself.

## Remediation

- Reissue the secure token in Github
- Add this token to environment variables in Appveyor: https://ci.appveyor.com/project/cncjs/cncjs/settings/environment (guessing at the URL here)
- Convert the token usage in appveyor.yaml to an environment variable matching the name set in Appveyor env vars

## Questions

- Is this token actually needed if Appveyor is installed as a Github app?

## Next steps

- Waiting 40m for the build to finish to see if status is reported properly on https://github.com/emcniece/cncjs/pull/1

emcniece, 2022-03-22:
Error behaviour replicated on https://github.com/emcniece/cncjs/pull/1

emcniece, 2022-03-22:
After some time, the initial build for https://github.com/emcniece/cncjs/pull/1 succeeded - I don't think this is related to an unauthorized deploy token.

Looking at the Github statuses published by Appveyor, `continuous-integration/appveyor/pr` builds the pull request and `continuous-integration/appveyor/branch` builds `master` branch. These builds are triggered each time a pull request is updated. The builds take about an hour, and under Appveyor's free tier the project is only allowed one build at a time. This creates quite a long queue for projects with many branches, pull requests, and commits.

The long queues are an inconvenience but for this problem they are a red herring - it is the latest branch builds that must be fixed, so that Appveyor can report a success on all open PRs.

It is strange that the in-progress statuses don't include a link to the active Appveyor build, something must have gone wrong with how Appveyor updates specific statuses. The statuses in question are all for rebuilding `master` branch. The Appveyor project history may have some information, we'll have to dig through the list to find a way to relate the build to a PR.

Next I will test a fresh CI build with a no-op PR.

emcniece, 2022-03-22:
Interesting - in https://github.com/emcniece/cncjs/pull/1 both builds (`/branch` and `/pr`) were queued immediately, and both had links to Appveyor. In https://github.com/cncjs/cncjs/pull/748 only the `/pr` build gets a link, and https://ci.appveyor.com/project/cheton/cncjs/history shows no intent to add the `/branch` build to the queue.

I suspect either the Appveyor project settings have been altered to create this behaviour, or there is some plan limitation the project is hitting.

@MitchBradley can you help confirm the settings at https://ci.appveyor.com/project/cheton/cncjs/settings? I'll post screenshots of my successful config:

Section 1:
[image]

Section 2:
[image]

Section 3:
[image]

MitchBradley, 2022-03-22:
I don't seem to have permissions on that Appveyor account.  My member status on the cncjs repo does not seem to extend to the appveyor setup.  I will try to get @Cheton involved.  He is in Taiwan so it might be awhile before he is online.

emcniece, 2022-03-23:
Thanks @MitchBradley. I wonder if you could check one more thing:

Is "continuous-integration/appveyor/branch" set as an expected check in the branch protection settings for `master` branch at https://github.com/cncjs/cncjs/settings/branches ?

cheton, 2022-03-23:
Hi @emcniece @MitchBradley 

I just removed the status check for `continuous-integration/appveyor/branch` from the master branch. Now you won't be blocked by the unwanted status check.

emcniece, 2022-03-23:
Thank you @cheton this is great!

While your attention is on the subject, can I ask what the vision for CI is for this project? I see many different platforms and status checks implemented in the PRs. I'm wondering if you're open to consolidating and improving these pipelines and the delivery flow.

cheton, 2022-03-23:
I switched from Travis CI to Appveyor and CircleCI because Travis CI had switched to a paid model and it's not friendly for OSS projects, especially when you have to make macOS builds.

For this repo, I use CircleCI and Appveyor for different purposes:

* The CircleCI configuration (https://github.com/cncjs/cncjs/blob/master/.circleci/config.yml) will run a workflow only when a tag is created. Its purpose is to build and deploy Docker images to Docker Hub (https://hub.docker.com/r/cncjs/cncjs/tags).

* The Appveyor (https://github.com/cncjs/cncjs/blob/master/appveyor.yml) configuration will trigger PR & master branch builds for Linux, macOS, and Windows environments. The installer artifacts will be created in the CI stage, but the assets will be uploaded to **GitHub Releases** only when the master branch gets updated or a tag release build was created.

  [image]

emcniece, 2022-03-23:
Thanks for the brief. Marking the issue as closed now that we've figured out the mystery status!

closed 2022-03-23 by emcniece

cheton, 2022-03-25:
Hi @emcniece 

@MitchBradley told me that you would like to help with CNCjs maintenance. I just sent you an invitation to join the cncjs-dev team with a "Maintain" role for the cncjs repo. Thanks for you assistance.

emcniece, 2022-03-25:
That's great, thank you @cheton! I'll introduce myself via email.