cncjs/cncjs #751: [Feature request] Warn before starting a Gcode that has no M3/M4

dwery, 2022-03-26:
As per title, it would be nice to have an (optional?) warning when pressing "play" to start executing a Gcode that has no M3/M4 command

neilferreri, 2022-04-01:
This should be up to the CAM program.  There are valid reasons why users may not want or need spindle control commands in the gcode.

Fusseldieb, 2022-05-09:
As a person who deployed CNCjs into various production environments and saw people breaking CNC tools several times (especially when the slicer is new and partially unconfigured), my suggestion would be adding a optional checkbox for enabling this kind of warning in the settings (opt-out).

risacher, 2022-05-22:
I use my 3018 cnc in four different ways: as a router, laser cutter, plotter, and drag knife (i.e. vinyl cutter).  When used as a plotter or drag knife, I don't need any M3/M4 commands. That said, it wouldn't be a big deal if there was a warning.