cncjs/cncjs #754: Watch Directory does not work with Windows App

MitchBradley, 2022-03-30:
First, the documentation does not explain how to set up a watch directory when you are using the Windows Electron App.  I knoiw two ways to set it up, neither of which are friendly enough for the typical "install and click" user that the app targets:

- You can edit the shortcut that starts the app to add "--watch-directory \<path\>" to the Target value.  Teaching average users to find and edit shortcuts is quite tedious.
- You can edit %userprofile%\.cncrc to add a "watchDirectory": "C:/Users/USER/path", entry.  That too is hard to explain and error prone, because people often do not know where their home directory is located, use editors that mess up plain text, make punctuation mistakes in the entry, get the comma wrong at the end of the JSON line, or make mistakes with / vs \\.

But even after you successfully set up the watch directory, it does not work.  If you browse the watch directory, you get an empty list of files.  The problem is this code in src/server/services/monitor/index.js:
```
    if (!root || pattern.indexOf(root) !== 0) {
        return [];
    }
```
The test `pattern.indexOf(root) !-- 0` fails because _pattern_ contains \ delimiters, while _root_ has / delimiters, so they do not match.

It can be fixed by this patch:
```
diff --git a/src/server/services/monitor/index.js b/src/server/services/monitor/index.js
index 084442ce..b8fc0d3e 100644
--- a/src/server/services/monitor/index.js
+++ b/src/server/services/monitor/index.js
@@ -14,7 +14,7 @@ const stop = () => {
 };
 
 const getFiles = (searchPath) => {
-    const root = monitor.root;
+    const root = path.normalize(monitor.root);
     const files = Object.keys(monitor.files);
     const pattern = path.join(root, searchPath, '*');
 
```

MitchBradley, 2022-03-30:
For people who cannot wait for a new release, here is a way to patch the electron app:

Using a plain text editor like Notepad, edit the file C:\Users\YOURNAME\AppData/Local/Programs/cncjs-app/resources/app/server/index.js .

Search for the line
```
  var root = monitor.root;
```
and replace it with
```
  var root = external_path_default.a.normalize(monitor.root);
```
After restarting CNCjs, it should be able to browse the watch directory, assuming that you have set it up with one of the two methods shown above.

MitchBradley, 2022-03-30:
Ideally, the setting pane would provide a way to setup the watch directory, which would ease the documentation problem, but that is a not-entirely-trivial coding task.

cheton, 2022-03-31:
I will publish a patch release for recent changes.

commit eeaf153e (2022-03-30): Normalize watch directory path to fix #754
closed 2022-03-31 by cheton