cncjs/cncjs #757: Windows Desktop App Does Not Start

Yarkinco, 2022-04-08:
#### Description

When I start windows desktop app it just seems to work on background, but it does not open at all. It doesn't matter if I start in a few times more, start with admin rights, compatibility mode etc. I installed it from 

https://github.com/cncjs/cncjs/releases/download/v1.9.24/cncjs-app-1.9.24-windows-x64.exe

#### Versions

- CNCjs: 1.9.24
- Node.js: 12.9.1
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

[image]

neilferreri, 2022-04-08:
Did it ever run?
Tried restarting the PC?

Yarkinco, 2022-04-08:
Yes. A few months ago it was working. I realized today, that it is not working. Uninstalled it, reinstalled latest version. Still not working.

Taruuk, 2022-07-02:
I have a similar issue. My program will run but only in background, no GUI. 

It seems to always display the CNCjs window on another screen. When I minimize/maximize from the taskbar a small animation appears which suggests the window is displayed on the imaginary screen to the left. No efforts of mine can move the window over. WIN+ARROWKEY or SHIFT+WIN+ARROWKEY will not move window.....

nickk2992, 2022-09-11:
did anyone find the fix to this?

MitchBradley, 2022-09-11:
Try all the techniques in [this link from Microsoft](https://support.microsoft.com/en-us/topic/some-programs-are-displayed-off-the-screen-b2a915b7-4705-c455-febf-8ddb6e7f0ae4) .  For Method 1, it might help to try SHIFT right-click before Maximize.

Asking "did anybody (here) find a fix" is unlikely to help you because it is a generic Windows problem that is not CNCjs-specific.  The Internet has a lot of suggestions for recovering hidden windows.

MitchBradley, 2022-09-11:
It can help to use Windows Display Settings to disconnect all displays except the main one.  When you do that, Windows tends to try to get all running applications onto the main display.  Good luck.

nickk2992, 2022-09-11:
I cannot switch to the app using alt-tab but it does show up as "running" in the Windows Task Manager.  Only one display connected

MitchBradley, 2022-09-11:
You could try killing cncjs, then deleting .cncrc and .cncjs-sessions from your home directory C:\Users\YOURNAME\ .  The idea being that maybe there is some resource specified in those files that the process is waiting for.

MitchBradley, 2022-09-11:
You could try Process Explorer to see if it gives you useful information.  Process Explorer is a Microsoft tool that is like a supercharged Task Manager.  On my system, there are three CNCjs processes.  Process Explorer shows their command lines.  The top level process is the main one that is started when you click on the CNCjs icon.  The other two have command lines that include "-type=renderer" and "-type=gpu-process".  It is the "gpu-process" one that causes a window to appear.  If I kill that one, the window disappears and then the process restarts and the window reappears.  If I kill the "renderer" process, the window stays up but become blank inside, and the renderer process does not automatically restart.  So I wonder if there is something about your GPU that CNCjs does not like.  Maybe a driver got updated or something.  Just guessing.

Yarkinco, 2022-09-11:
@nickk2992 not yet. After trying to fix it for a long time, I gave up and started to use UGS, universal gcode sender. Until today.

Still I could not able to fix it but @MitchBradley 's suggestion might lead us something. Unfortunately, first two suggestion did not work for me. But, Process Explorer show me only 2 cncjs.exe. Main one and the one with "-type=gpu-process". There isn't the third one with renderer command line. I think this is the reason of our problems. Now we need to find a way to fix it.

Taruuk, 2022-09-11:
> 
 SHIFT --> Right Click --> Maximize 

This just solved a problem I've been stumped by for almost an entire year.  Thank you @MitchBradley !

cheton, 2023-01-19:
If you are experiencing issues with the CNCjs Electron app on Windows, try following the troubleshooting steps outlined in the wiki: https://github.com/cncjs/cncjs/wiki/Troubleshooting#troubleshoot-electron-app-on-windows.

If an error occurs, an error message will appear in the command prompt.

I will close this issue. You can update here if you discover any new information.

closed 2023-01-19 by cheton