cncjs/cncjs #759: Jog does not function: Marlin

rflulling, 2022-04-20:
#### Description

Jog does not function. 
Jog works find in other products such as S3D and CURA. 
Does nothing in CNCjs. 
Serial log indicates that CNCjs is not even trying to send a command. 
I see an older bug ticket indicates this function was FIXED?

#### Versions

- CNCjs: 1.9.24 DESKTOP
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [x] MARLIN 2.0.9.2
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device
- [X] SKR PRO 1.1

#### Operating System

- [ ] Not Applicable
- [x] 64bit Windows 10 Pro N 
- [ ] Mac
- [ ] Linux

rflulling, 2022-04-20:
Edit: 
Home command is generated ONLY if I hit the blue button on the upper right. 
[image]

Nothing on the Axis Widget causes command to be sent.
Home and Zero on the Widget do nothing. Keyboard commands do nothing.

rflulling, 2022-06-05:
Have recently tested CNCjs with toy CNC running GRBL and the jog functions work fine. 

To be clear, when I say the jog works, I am not just looking at the machine. I am looking at the console. 

With GRBL, the job commands directly generate code to the console window. However for reasons unknown, this NEVER happens when connected to Marlin. However other actions such as home both result in code being generated and sent on the console window.

rflulling, 2022-06-05:
If there is a suggestion as to a dependency that should be installed for the desktop program to work, I am open to suggestions. 

I have tried to install through NPM and although much did install. Trying to add CNCjs resulted in more errors than I can shake a stick at. Mostly it seemed about outdated components that had been replaced or externally updated on their own.

rflulling, 2022-08-06:
Installed docker... there is no way that thing is not a massive wide open door for hackers... disabled when not in use. 
Could not get CNCjs to install. It installed then acted like it could not find the files. 

Tried rerunning base installers with both 32bit java and 64 bit Java. Something about this system just doesn't like the program. Maybe security is doing something. 

Updated the Desktop version. No issues. Runs fine for GRBL. But Marlin, nope. Jog tools are useless. They don't even try to generate code, or send it. I am watching over the serial log. But on GRBL, no issues, other than GRBL itself. 

---

I would really like for this software to work. I don't even mind a small sponsorship, especially if I can get the machines running the way I want and it earns some cash. Not supporting those I depend on to keep the business running would be nuts. 

Am I nuts to hope to see this fixed?

cheton, 2022-08-06:
Note that you have to set baud rate to 25000 to allow Marlin to connect. I can't see any clue from your provided information. 

I suggest you provide server logs and operation videos/screenshots that how you operate the app.

Please go to the "Troubleshoot Electron app on Windows" section in below link to learn how to dump server logs:
https://github.com/cncjs/cncjs/wiki/Troubleshooting#troubleshoot-electron-app-on-windows

For CNCjs 1.9.26, you have to download Electron v4:
https://github.com/electron/electron/releases/download/v4.2.12/electron-v4.2.12-win32-x64.zip

Here are the steps:
1. Extract "electron-v4.2.12-win32-x64.zip" to a temp folder (e.g., `d:\Temp\electron-v4.2.12-win32-x64`)
2. Go to the CNCjs app folder (e.g., `C:\Users\cheton\AppData\Local\Programs\cncjs-app\resources\app` as mine) and run `d:\Temp\electron-v4.2.12-win32-x64\electron.exe main.js -vvv` to launch the app. Then you will see the server logs for further estimation.

[image]

rflulling, 2022-08-11:
Marlin is configured to connect at 115200 so unless the CNCjs has a limit that prevents this. Marlin connects just fine. That is not the issue. The issue is that CNCjs jog does nothing at all and I have come across other that gave up waiting on a fix. I haven't given up because I think that CNCjs has the most potential of most of the products I have seen. Second only to Repetier Host/Server but they lack any tools for laser or router. 

So just got the the verbose thing to work one time, via power shell. Found out if you try to run the command in Command Prompt, even with admin, it says "This app can't run on your PC. Find a version that can run on your PC, check with the software publisher".  Funny considering it just did and in most other cases windows will allow the over ride. Command Prompt, it's just locked. Microsoft is such a mindless mess of headaches. So the only issue is in getting a window open I can copy from. 
Power Shell doesn't want to let me select all Copy, or save as. 

So the only issue is in getting a window open I can copy from. 
Power Shell doesn't want to let me select all Copy, or save as. 

Ohh, either Electron delivered a payload and self destructed or something else nuked it. The app says its 0k out of 89mb when installed. That's a very curious happening. 

Restored the files... Trying again. 

Ok got it working in command prompt. 
Opened program. Sent home. Then tried to toggle and run every combination in the Jog tool. keyboard, manual data input, none. Pressed every button and every switch, watching to see if anything would generate a message in serial or make the machine move. Nope. "Home" is the only thing that works.  

I am guessing I can CNC a file as things are. But I am not testing that until the basics work. 

[Electron_CNCjs_verbose.txt](https://github.com/cncjs/cncjs/files/9305471/Electron_CNCjs_verbose.txt)

rflulling, 2022-08-12:
So the next question, am I supposed to physically install something to onto the Marlin based controller? It seems that for say other devices this is true. However since you don't really add programs to marlin this would seem to be false. Devices like Raspberry Pi and Beagle Bone Black have a dedicated OS running and programs can be added. But others like Arduino, ESP32 and STM32, must be included as part of the sketch to be compiled and uploaded as a whole. I know there are some exceptions to that. But as far as I know, Marlin is not setup to open programs, it is the only program, right? 

I see the code trying to reach out to send HTML and other various encoded commands. Marlin is like, I know not what you speak and ignores the commands without question. By the way are not sent over the serial terminal. So i suspect they are being sent as IP?! Anything sent over serial that is a valid command is acted upon and or buffered for execution. ie Home, which works. But Jog sends these weird commands. 

My log is filled with these:
debug service:cncengine socket.command("COM3", "gcode"): id=u_uXeuDZe-I-6g70AAAA     at Socket.<anonymous> (C:\Users\SystemG\AppData\Local\Programs\cncjs-app\resources\app\server\services\cncengine\CNCEngine.js:422:15)

These entries are being poking jog buttons and switching functions. Hopeful something will work and the machine will move. 
But if the command is being sent as IP, then it will never work. Also why in the world, when connected over USB is it trying to connect over IP?!!! There is literally no user accessible function to make that happen, or disable it. 

Gotta stop here to install an update that Microshaft decided is a requirement. Its setup broke my computer at work cause programs to refuse to launch. Then cause really weird lockups after reboot. Here at home the programs are refusing to launch. 
-Here is to hoping I don't have to wipe the whole drive and start over again thanks to a non optional update,.

cheton, 2022-08-12:
My testing board was running Marlin 1.1.8 and it's working well when connected with 250000 baud rate. 

Otherwise, I see a major difference regarding the Marlin firmware version. I will check later if CNCjs is still compatible with Marlin 2.0.x.

[image]

```
2022-08-12T05:05:40.107Z - debug service:cncengine socket.open("COM8", {"controllerType":"Marlin","baudrate":250000,"rtscts":false}): id=sTCMdqBN-h_wUAxrAAAA     at Socket.<anonymous> (C:\Users\cheton\AppData\Local\Programs\cncjs-app\resources\app\server\services\cncengine\CNCEngine.js:324:15)
2022-08-12T05:05:40.114Z - debug controller:Marlin Add socket connection: id=sTCMdqBN-h_wUAxrAAAA     at MarlinController.addConnection (C:\Users\cheton\AppData\Local\Programs\cncjs-app\resources\app\server\controllers\Marlin\MarlinController.js:1140:11)
2022-08-12T05:05:40.126Z - debug controller:Marlin Connected to serial port "COM8"     at SerialPort.<anonymous> (C:\Users\cheton\AppData\Local\Programs\cncjs-app\resources\app\server\controllers\Marlin\MarlinController.js:1074:13)
2022-08-12T05:05:41.080Z - silly controller:Marlin < start
2022-08-12T05:05:41.081Z - silly controller:Marlin < echo:Marlin 1.1.8
2022-08-12T05:05:41.082Z - silly controller:Marlin < echo: Last Updated: 2017-12-25 12:00 | Author: (none, default config)
2022-08-12T05:05:41.082Z - silly controller:Marlin < echo:Compiled: Jun 29 2018
2022-08-12T05:05:41.082Z - silly controller:Marlin < echo: Free Memory: 5539  PlannerBufferBytes: 1264
2022-08-12T05:05:41.082Z - silly controller:Marlin < echo:Hardcoded Default Settings Loaded
2022-08-12T05:05:41.083Z - silly controller:Marlin < echo:  G21    ; Units in mm
2022-08-12T05:05:41.083Z - silly controller:Marlin < echo:Filament settings: Disabled
2022-08-12T05:05:41.083Z - silly controller:Marlin < echo:  M200 D3.00
2022-08-12T05:05:41.083Z - silly controller:Marlin < echo:  M200 D0
2022-08-12T05:05:41.083Z - silly controller:Marlin < echo:Steps per unit:
2022-08-12T05:05:41.083Z - silly controller:Marlin < echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
2022-08-12T05:05:41.083Z - silly controller:Marlin < echo:Maximum feedrates (units/s):
2022-08-12T05:05:41.083Z - silly controller:Marlin < echo:  M203 X300.00 Y300.00 Z5.00 E25.00
2022-08-12T05:05:41.084Z - silly controller:Marlin < echo:Maximum Acceleration (units/s2):
2022-08-12T05:05:41.084Z - silly controller:Marlin < echo:  M201 X3000 Y3000 Z100 E10000
2022-08-12T05:05:41.084Z - silly controller:Marlin < echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
2022-08-12T05:05:41.084Z - silly controller:Marlin < echo:  M204 P3000.00 R3000.00 T3000.00
2022-08-12T05:05:41.084Z - silly controller:Marlin < echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
2022-08-12T05:05:41.084Z - silly controller:Marlin < echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
2022-08-12T05:05:41.084Z - silly controller:Marlin < echo:Home offset:
2022-08-12T05:05:41.084Z - silly controller:Marlin < echo:  M206 X0.00 Y0.00 Z0.00
2022-08-12T05:05:41.084Z - silly controller:Marlin < echo:PID settings:
2022-08-12T05:05:41.085Z - silly controller:Marlin < echo:  M301 P22.20 I1.08 D114.00
2022-08-12T05:05:41.355Z - silly controller:Marlin < FIRMWARE_NAME:Marlin 1.1.8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
2022-08-12T05:05:41.355Z - silly controller:Marlin < Cap:SERIAL_XON_XOFF:0
2022-08-12T05:05:41.356Z - silly controller:Marlin < Cap:EEPROM:0
2022-08-12T05:05:41.356Z - silly controller:Marlin < Cap:VOLUMETRIC:1
2022-08-12T05:05:41.356Z - silly controller:Marlin < Cap:AUTOREPORT_TEMP:1
2022-08-12T05:05:41.356Z - silly controller:Marlin < Cap:PROGRESS:0
2022-08-12T05:05:41.356Z - silly controller:Marlin < Cap:PRINT_JOB:1
2022-08-12T05:05:41.356Z - silly controller:Marlin < Cap:AUTOLEVEL:0
2022-08-12T05:05:41.357Z - silly controller:Marlin < Cap:Z_PROBE:0
2022-08-12T05:05:41.357Z - silly controller:Marlin < Cap:LEVELING_DATA:0
2022-08-12T05:05:41.357Z - silly controller:Marlin < Cap:BUILD_PERCENT:0
2022-08-12T05:05:41.357Z - silly controller:Marlin < Cap:SOFTWARE_POWER:0
2022-08-12T05:05:41.357Z - silly controller:Marlin < Cap:TOGGLE_LIGHTS:0
2022-08-12T05:05:41.357Z - silly controller:Marlin < Cap:CASE_LIGHT_BRIGHTNESS:0
2022-08-12T05:05:41.357Z - silly controller:Marlin < Cap:EMERGENCY_PARSER:0
2022-08-12T05:05:41.357Z - silly controller:Marlin < ok
2022-08-12T05:05:41.357Z - silly controller:Marlin controller.on('ok'): source=server, line="M115", res={"raw":"ok"}
2022-08-12T05:05:41.387Z - silly controller:Marlin < X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
2022-08-12T05:05:41.387Z - silly controller:Marlin controller.on('pos'): source=server, line="M114", res={"pos":{"x":"0.00","y":"0.00","z":"0.00","e":"0.00"},"raw":"X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0"}
2022-08-12T05:05:41.387Z - silly controller:Marlin < ok
2022-08-12T05:05:41.388Z - silly controller:Marlin controller.on('ok'): source=server, line="M114", res={"raw":"ok"}
2022-08-12T05:05:41.898Z - silly controller:Marlin < X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
2022-08-12T05:05:41.898Z - silly controller:Marlin controller.on('pos'): source=server, line="M114", res={"pos":{"x":"0.00","y":"0.00","z":"0.00","e":"0.00"},"raw":"X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0"}
[... cut ...]
2022-08-12T05:05:53.469Z - silly controller:Marlin controller.on('temperature'): source=server, line="M105", res={"ok":true,"extruder":{"deg":"-15.00","degTarget":"0.00","power":"0"},"heatedBed":{},"raw":"ok T:-15.00 /0.00 @:0"}
2022-08-12T05:05:53.470Z - silly controller:Marlin controller.on('ok'): source=server, line="M105", res=undefined
2022-08-12T05:05:53.638Z - debug service:cncengine socket.close("COM8"): id=sTCMdqBN-h_wUAxrAAAA     at Socket.<anonymous> (C:\Users\cheton\AppData\Local\Programs\cncjs-app\resources\app\server\services\cncengine\CNCEngine.js:396:15)
```

cheton, 2022-08-14:
@rflulling I think I found a problem that might cause this **No Response** issue.

Here is the code (in CNCjs) for parsing the temperature result:

https://github.com/cncjs/cncjs/blob/master/src/server/controllers/Marlin/MarlinLineParserResultTemperature.js

```js
class MarlinLineParserResultTemperature {
    // ok T:0
    // ok T:293.0 /0.0 B:25.9 /0.0 @:0 B@:0
    // ok T:293.0 /0.0 B:25.9 /0.0 T0:293.0 /0.0 T1:100.0 /0.0 @:0 B@:0 @0:0 @1:0
    // ok T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0
    // ok T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0 W:?
    // ok T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0 W:0
    //  T:293.0 /0.0 B:25.9 /0.0 @:0 B@:0
    //  T:293.0 /0.0 B:25.9 /0.0 T0:293.0 /0.0 T1:100.0 /0.0 @:0 B@:0 @0:0 @1:0
    //  T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0
    static parse(line) {
        let r = line.match(/^(ok)?\s+T:[0-9\.\-]+/i);
        if (!r) {
            return null;
        }

        const payload = {
            ok: line.startsWith('ok'),
            extruder: {},
            heatedBed: {}
        };

        const re = /(?:(?:(T|B|T\d+):([0-9\.\-]+)\s+\/([0-9\.\-]+)(?:\s+\((?:[0-9\.\-]+)\))?)|(?:(@|B@|@\d+):([0-9\.\-]+))|(?:(W):(\?|[0-9]+)))/ig;
```

And this is the response message recorded from your Marlin 2.0.9.2:

```
ok T0:27.58 /0.00 B:28.27 /0.00 T0:27.58 /0.00 T1:27.37 /0.00 @:0 B@:0 @0:0 @1:0
 T0:27.72 /0.00 B:28.38 /0.00 T0:27.72 /0.00 T1:27.28 /0.00 @:0 B@:0 @0:0 @1:0
wait
wait
 T0:27.66 /0.00 B:28.30 /0.00 T0:27.66 /0.00 T1:27.28 /0.00 @:0 B@:0 @0:0 @1:0
```

It appeared that the regular expression assume the line should begin with either `ok T:` or ` T:`. 

```js
let r = line.match(/^(ok)?\s+T:[0-9\.\-]+/i);
if (!r) {
  return null;
}
```

However, Marlin 2.0.x reported `ok T0:` and ` T0:` rather than `ok T:` and ` T:`, so CNCjs ignored this command.

I will try to provide and hot fix and make a patch release.

---

Note: If you're familiar with JavaScript programming and regular expression, you can try to modify the source code, that will save your time for further verification. You can find the file at `C:\Users\SystemG\AppData\Local\Programs\cncjs-app\resources\app\server\controllers\Marlin\MarlinLineParserResultTemperature.js` for a quick fix as below:

```diff
     //  T:293.0 /0.0 B:25.9 /0.0 T0:293.0 /0.0 T1:100.0 /0.0 @:0 B@:0 @0:0 @1:0
     //  T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0
     static parse(line) {
-        let r = line.match(/^(ok)?\s+T:[0-9\.\-]+/i);
+        let r = line.match(/^(ok)?\s+T[0-9]*:[0-9\.\-]+/i);
         if (!r) {
             return null;
         }
```

cheton, 2022-08-14:
@rflulling There's another parsing issue at https://github.com/cncjs/cncjs/issues/581 (PR #583) that CNCjs 1.9.21 and earlier versions cannot make jog working because it failed to parse temperature report ` T:0` if there is no temperature sensor.

> `T:0` is the result of `M105` if `HAS_TEMP_SENSOR` is `false`

rflulling, 2022-08-14:
Thank you for taking the time to look into this. I would have never guessed from the logs that CNCjs was getting hung up on temp sensor report. 

If you feel the issue is with marlin and that it is not generating a output in a way that is considered industry standard I can also file a ticked on their side and see if they will agree. 

Please keep in mind that with how expansive Marlin 2.0.x is, if CNCjs is dependent on a temp sense from Marlin, then it will need to be flexible enough to understand that Marlin may have 0 or many extruders. I will have to look to see how many it now supports. This rapid expansion is due to new boards that allow easily 8+ extruders and 8+ motors. I think that CNCjs knows enough to expect one of two formats and maybe ask user or check to see how many extruders there from firmware, should work well to configure it's expectations. I think since my last update Marlin has only added even more tools, and even more flexibility to support just about any configuration of workbench machine that might be Needed. 

Unfortunately while, I can and will gather any info I can and report back. My programing is limited to BASIC, HTML, C and C++. Java is a foreign animals to me. I haven't done basic since High school and my use of C or C++ is enough to be dangerous, I can read it, but using it well enough draft my own code is, extremely difficult. So, I think I will not be able to offer much support in coding. Unless the work is in C or C++ and I happen to recognize an error. 

---

My intent is to use CNCjs to drive Marlin based controllers for CNC work. 

My home test rig is a 3D printer running on a SKR PRO v1.1, so right now it is set up with dual extruders, but I would like to be be able to later configure it to exist with 3 extruders, a router, and a laser, maybe more. If CNCjs can be the desktop controller to setup, test and manually adjust prior to file execute then it's going to fill a very large roll. -Currently S3D is the only desktop controller I use, but it has no easy way of switching CNC or Laser relays other than by script, and its optimized for FDM not Reduction (CNC).

Long term I intend to build new machines both CNC, laser and 3D. I also have two more commercial bench top machines using proprietary 16bit controllers. I look forward to giving them the 32bit TMC driver treatment. 

I know I am going to step on toes. But those toes are long over due being stepped on and I really believe that CNCjs driving Marlin will be the tools for the job. :-)

closed 2022-08-18 by cheton