cncjs/cncjs #760: Comments on M0/M6 command not working in gcode

wb9tpg, 2022-05-17:
#### Description

I've manually created a simple gcode file

G0 X0Y0
M0;Say Something
G0 X10Y10 

The M command comment does **not display** when running the gcode from a file but it will display if I place the same gcode in a macro.  I seems it's supposed to work in both test cases based in the description in the release.  

#### Versions

- CNCjs: 1.9.25
- Node.js: 10.24.1
- NPM: 6.14.12

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

Bart95580, 2022-05-28:
same issue for me.
It's not working from gcode file.
so, it's not useful :-(

closed 2022-06-22 by cheton

cheton, 2022-06-22:
Published [v1.9.26](https://github.com/cncjs/cncjs/releases/tag/v1.9.26)

Bart95580, 2022-06-22:
Thank you very much
My first test with M06 command is successful

wb9tpg, 2022-07-29:
Tested this today and it passed all my tests.  Thanks for the fix